coolsnowwolf/lede #8472: easymesh编译报错

ThermalEng, 2021-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：easymesh action编译报错

### (2) 路由器型号和固件版本

A：ax3600

### (3) 详细日志

A：* check_conflicts_for: 	wpad-mesh-openssl * 
Package batctl-default (2021.4-1) installed in root is up to date.
 * opkg_install_cmd: Cannot install package luci-app-easymesh.
 * check_conflicts_for: The following packages conflict with wpad-openssl:
 * check_conflicts_for: 	wpad-mesh-openssl * 
 * opkg_install_cmd: Cannot install package wpad-openssl.
make[2]: *** [package/Makefile:70: package/install] Error 255

WYC-2020, 2021-12-18:
冲突吧

ThermalEng, 2021-12-18:
知道是冲突，但都是默认的包。还是要请解决下的

ahhnz, 2021-12-23:
网络里关闭wpad选项network-->wpad

coolsnowwolf, 2026-09-06:
日志显示的是 wpad-mesh-openssl 与 wpad-openssl 的互斥依赖，属于同时选择冲突的无线实现，并非 easymesh 源码编译错误。应在配置中二选一，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf