coolsnowwolf/lede #8474: OpenWRT的IPSec客户端连上VPN后无法让LAN口的电脑走VPN流量

hunhun010164, 2021-12-18:
**硬件:** TPLink TL-MR11U,换了32M ROM和64M RAM
**固件:** 因18之后不再支持,所以只能刷了官方最后一版的17.01.7, r4030-6028f00df0
因ROM还是较小,所以只能安装最基本的软件: openssl-util strongswan-minimal strongswan-mod-curve25519 strongswan-mod-af-alg strongswan-mod-openssl strongswan-mod-pem strongswan-mod-eap-identity strongswan-mod-eap-mschapv2

因只有一个网口,所以无线同时做无线客户端和AP,网口做LAN
**/etc/config/wireless:**
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'platform/ar933x_wmac'
	option htmode 'HT20'
	option disabled '0'
	option channel '6'
	option country 'US'

config wifi-iface
	option network 'wan'
	option ssid 'test'
	option device 'radio0'
	option mode 'sta'
	option key '12345678'
	option encryption 'psk2+ccmp'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'AP'
	option encryption 'psk2+ccmp'
	option key '12345678'

**/etc/config/network:**
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd8b:0835:56e3::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0'
	option proto 'static'
	option ipaddr '192.168.100.1'
	option netmask '255.255.255.0'
	list dns '8.8.8.8'
	list dns '8.8.4.4'
	option ip6assign '60'

config interface 'wan'
	option proto 'dhcp'

**/etc/config/firewall的主要部分:**
config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option network 'lan'

config zone
	option name 'wan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option masq '0'
	option mtu_fix '1'
	option network 'wan'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Dropbear'
	option src 'wan'
	option proto 'tcp'
	option dest_port '22'
	option target 'ACCEPT'

**ipsec.conf:**
config setup
	uniqueids = no

conn SH
	keyexchange=ikev2
	right=my.domain
	rightsubnet=0.0.0.0/0
	leftsourceip=%config
	leftsendcert=never
	leftauth=eap-mschapv2
	eap_identity=test
	auto=route

**现在的情况是:**
路由器可以连上VPN服务器,能解析地址,能ping通外网,检查traceroute也是走的VPN服务器,非WAN,满意.

**问题是:**
只有ipsec stop的情况下,下挂的电脑,无论是连LAN口还是连无线,都可以拿到192.168.100.X的IP,且能上网
一旦ipsec start了,下挂的电脑,无论是连LAN口还是连无线,虽然IP还在,但是就是不能上网,更无法走VPN的流量.

**需要解决的:**
路由器连上VPN后,能让下挂的电脑的所有流量统统走VPN.谢谢各位大神

1303672189, 2021-12-24:
遇到了类似的问题，期待可以解决

wzh4809, 2021-12-25:
防火墙-转发 选择接受

hunhun010164, 2021-12-25:
> 防火墙-转发 选择接受

能否具体一点呢?我是小白.谢谢.

coolsnowwolf, 2026-09-06:
该问题基于 OpenWrt 17.01.7（2018 年旧版本）和已停止支持的 TL-MR11U，配置也未提供当前 LEDE 可复现环境。旧版 strongSwan 路由策略问题不属于当前仓库可验证缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf