coolsnowwolf/lede #8476: 二次编译出错了

mzl0408, 2021-12-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
今天二次编译一下，结果报错了，日志贴在下面了
Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：Configuring php7-mod-gd.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * check_conflicts_for: The following packages conflict with php8:
 * check_conflicts_for: 	php7 * 
 * opkg_install_cmd: Cannot install package luci-app-kodexplorer.
 * check_conflicts_for: The following packages conflict with php8:
 * check_conflicts_for: 	php7 * 
 * opkg_install_cmd: Cannot install package php8.
 * check_data_file_clashes: Package php8-cgi wants to install file /home/meng/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/bin/php-cgi
	But that file is already provided by package  * php7-cgi
 * opkg_install_cmd: Cannot install package php8-cgi.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/meng/lede'
make[1]: *** [package/Makefile:111: /home/meng/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/meng/lede'
make: *** [/home/meng/lede/include/toplevel.mk:230：world] 错误 2

boshi116, 2021-12-18:
一样的我问题

Boos4721, 2021-12-18:
**clean your config and remake menuconfig.**

loftyaspiration, 2021-12-18:
cd lede                                                   进入LEDE目录
git pull                                                  同步更新源码
./scripts/feeds update -a && ./scripts/feeds install -a   更新FEEDS
rm -rf ./tmp && rm -rf .config                        清除编译配置和缓存
make menuconfig                                         进入编译配置菜单
make V=s -j$(nproc)                                 多线程编译

清除一下编译配置和缓存

jxph1123, 2021-12-23:
重新拉源码重新来，啥毛病没了

closed 2021-12-28 by AmadeusGhost