coolsnowwolf/lede #8478: n2n 3.0上了，但是luci app还没跟上，luci UI端能优化否适配一下？

GreatMichaelLee, 2021-12-18:
1. 3.0的federation功能能否支持配置使能-I -J -P的用户名密码及密钥，以及federation名称能否在UI端的配置
2. 指定edge的MAC地址功能
3. federation node的select mac/select-rtt参数
4. 多supernode的'+'方式配置，目前写死了2个。
5. -A2-A5 加密方式参数配置

恳请大佬有空出手~十分感谢

GreatMichaelLee, 2021-12-18:
目前我是在/etc/init.d/n2n_v2 里手工加的这些参数，能work,但是回显不到luci web UI端。比较麻烦。

ccblinda, 2021-12-18:
n2n3.0后，luci插件废了！

mythsman, 2023-01-08:
系统 -> 软件包 -> 配置 -> 发行版软件源 中 ，luci 源更新下，然后就有最新的支持 3.0 的 luci-app-n2n 了。

不知道为啥原装固件的luci源是一个老版本。。。
```
src/gz openwrt_luci https://downloads.immortalwrt.org/releases/packages-18.06-k5.4/aarch64_generic/luci
```

改成新版本就可以了。
```
src/gz openwrt_luci https://downloads.immortalwrt.org/releases/21.02.1/packages/aarch64_generic/luci
```

coolsnowwolf, 2026-09-05:
n2n 3.0 当前包已提供，但 issue 请求的是 federation、MAC 选择、多 supernode、A2-A5 等一组新 UI 功能，属于功能需求而非明确缺陷；关闭为 not planned。

closed 2026-09-05 by coolsnowwolf