coolsnowwolf/lede #8480: 今天删完dnsproxy，luci-app-turboacc就。。。。

Jason6111, 2021-12-18:
WARNING: Makefile 'package/lean/luci-app-turboacc/Makefile' has a dependency on 'dnsproxy'

GreatMichaelLee, 2021-12-18:
一样问题，我没删dnsproxy，但是也报这个，还有另一个dependency issue,是不可言说插件里的navieproxy 说依赖ninja/host

WYC-2020, 2021-12-18:
我这边一切正常，你是feed加了那些，有问题吧

smallprogram, 2021-12-18:
一样
```
Configuring 喂吐ray-Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency dnsproxy for luci-app-turboacc
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-turboacc found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-turboacc-zh-cn:
 *      dnsproxy
 * opkg_install_cmd: Cannot install package luci-i18n-turboacc-zh-cn.
 ```

smallprogram, 2021-12-18:
#8443

smallprogram, 2021-12-18:
@Beginner-Go 老铁，修复一下呗~

minlang112, 2021-12-18:
移除helloworld重新git clone，然后make distclean，不会继续报错的

Beginner-Go, 2021-12-18:
> @Beginner-Go 老铁，修复一下呗~

依次执行
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a

closed 2021-12-18 by Jason6111