coolsnowwolf/lede #8482: Samba4编译报错

yaotthaha, 2021-12-19:
`Checking for openpty                                                              : ok
Checking for system installation of Python module markdown                        : not found
        Unable to find Python module 'markdown'. Please install the system package: python3-markdown'.

Makefile:449: recipe for target '/mnt/data/code/openwrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.7/.configured_8eafabf5ddcf6988c2cf2c669996767d' failed
make[3]: *** [/mnt/data/code/openwrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.7/.configured_8eafabf5ddcf6988c2cf2c669996767d] Error 1
make[3]: Leaving directory '/mnt/data/code/openwrt/openwrt/feeds/packages/net/samba4'
time: package/feeds/packages/samba4/compile#31.52#6.51#34.71
    ERROR: package/feeds/packages/samba4 failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/samba4/compile' failed
make[2]: *** [package/feeds/packages/samba4/compile] Error 1`

lxl6125, 2021-12-19:
Checking for system installation of Python module markdown : not found
Unable to find Python module 'markdown'. Please install the system package: python3-markdown'.

gili-gili, 2021-12-20:
**Unable to find Python module 'markdown'. Please install the system package: python3-markdown'.**

看一眼报错信息吧老哥（捂脸）

yaotthaha, 2021-12-21:
我知道啊，但是安装 python3-markdown 之后依然报错

Ra2-IFV, 2022-01-01:
> 我知道啊，但是安装 python3-markdown 之后依然报错

咋装的，直接apt install？

yaotthaha, 2022-01-02:
已解决，重新拉取重新编译即可

closed 2022-01-02 by yaotthaha