coolsnowwolf/lede #8483: netdata有问题

Karmylr, 2021-12-19:
[image]

xinb, 2021-12-19:
同问题

zhengyiquan, 2021-12-19:
手机显示就没问题，或主题问题？

zhengyiquan, 2021-12-19:
手机显示wu问题啊。或主题问题？

Karmylr, 2021-12-19:
> 手机显示wu问题啊。或主题问题？

都不是。应该是最新源码的bug

kevinCSDN, 2021-12-19:
同问题，编译了几次都是这样。偶尔正常，表盘还是会消失。

wkq2004, 2021-12-19:
刚打开正常一闪而过，就和一楼显示一样了。

CuteTony, 2021-12-20:
同问题

zhengyiquan, 2021-12-20:
前几天也是这样，昨晚从新编译了，没这个问题。用的是sirpdboy的netdata

zhoufeiyu2001, 2021-12-21:
同样也是这样子

WYC-2020, 2021-12-21:
我记得我说过解决方法的嘛

CuteTony, 2021-12-21:
> 我记得我说过解决方法的嘛

怎么解决呢

Karmylr, 2021-12-21:
> 我记得我说过解决方法的嘛

发个链接

ll26571, 2021-12-22:
同问题

kevinCSDN, 2021-12-22:
> 前几天也是这样，昨晚从新编译了，没这个问题。用的是sirpdboy的netdata

sirpdboy里的版本低，好像还是汉化的，可以替换掉lean中的应用luci-app-netdata再编译

dfbucy090, 2021-12-22:
> > 前几天也是这样，昨晚从新编译了，没这个问题。用的是sirpdboy的netdata
> 
> sirpdboy里的版本低，好像还是汉化的，可以替换掉lean中的应用luci-app-netdata再编译

请问，如何替换？本人小白，请写出执行指令脚本，不胜感激！

kevinCSDN, 2021-12-22:
> > > 前几天也是这样，昨晚从新编译了，没这个问题。用的是sirpdboy的netdata
> > 
> > 
> > sirpdboy里的版本低，好像还是汉化的，可以替换掉lean中的应用luci-app-netdata再编译
> 
> 请问，如何替换？本人小白，请写出执行指令脚本，不胜感激！

同是小白，请参考
git clone https://github.com/coolsnowwolf/lede
cd lede/package/lean
rm -rf luci-app-netdata
git clone https://github.com/sirpdboy/luci-app-netdata
cd ../..
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig

dfbucy090, 2021-12-23:
> 同是小白，请参考 git clone https://github.com/coolsnowwolf/lede cd lede/package/lean rm -rf luci-app-netdata git clone https://github.com/sirpdboy/luci-app-netdata cd ../.. ./scripts/feeds update -a [...]

谢谢您，很有用。

Michaol, 2021-12-26:
直接在界面卸载netdata，然后在可用软件包重装netdata 1.30.1-1，在编译目录\lede\bin\packages\x86_64\base找回luci-app-netdata_1.0-3_all.ipk和luci-i18n-netdata-zh-cn_1.0-3_all.ipk，直接文件传输上传，安装后就正常用回旧版本了。

CuteTony, 2021-12-30:
根本原因不在 luci-app-netdata，而是 [package仓库](https://github.com/coolsnowwolf/packages) 的 commit [ffab1a](https://github.com/coolsnowwolf/packages/commit/ffab1a44b15e90ce14d86b77873fe4399c562524) 中对 netdata 的更新，解决方法也很简单，暂时别用新的 1.31 版本就可以了。
你可以用之前编译好的 netdata 1.30.1-1，如果没有的话可以在 LEDE_DIR/feeds/packages 下撤销该次 commit，例如使用 `git revert ffab1a44b15e90ce14d86b77873fe4399c562524` 命令，然后重新 安装 feeds 并编译就好了。

CuteTony, 2021-12-31:
> 根本原因不在 luci-app-netdata，而是 [package仓库](https://github.com/coolsnowwolf/packages) 的 commit [ffab1a](https://github.com/coolsnowwolf/packages/commit/ffab1a44b15e90ce14d86b77873fe4399c562524) 中对 [...]

该仓库今日的 commit [2a1654](https://github.com/coolsnowwolf/packages/commit/2a1654b77a56f41dd1010b3f34c94f10a81c462a) 已经 revert commit [ffab1a](https://github.com/coolsnowwolf/packages/commit/ffab1a44b15e90ce14d86b77873fe4399c562524)，修复了这一错误。执行
`./scripts/feeds update -a`
`./scripts/feeds install -a`
拉取安装更新后，即可解决 netdata 问题。如果依旧出现闪一下的情况，清除浏览器缓存即可。

dfbucy090, 2021-12-31:
> 该仓库今日的 commit [2a1654](https://github.com/coolsnowwolf/packages/commit/2a1654b77a56f41dd1010b3f34c94f10a81c462a) 已经 revert commit [...]

32位编译出来还是不行，netdata显示一半字体或者黑屏。

closed 2022-02-12 by Beginner-Go