coolsnowwolf/lede #8484: [HELP]Compile Error编译错误

gili-gili, 2021-12-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败。次次编译错误都不一样，属实给我整无语了

### (2) 路由器型号和固件版本

A：Friendly Elec. NanoPi R4S

### (3) 详细日志

Packaged contents of /home/ubuntu/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-ieee802154-6lowpan into /home/ubuntu/lede/bin/targets/rockchip/armv8/packages/kmod-ieee802154-6lowpan_5.10.87-1_aarch64_generic.ipk
ERROR: module '/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.87/drivers/media/cec/cec.ko' is missing.
make[3]: *** [/home/ubuntu/lede/target/linux/rockchip/modules.mk:40: /home/ubuntu/lede/bin/targets/rockchip/armv8/packages/kmod-drm-rockchip_5.10.87-1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/ubuntu/lede/package/kernel/linux'
time: package/kernel/linux/compile#74.43#23.36#80.95
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:230: world] Error 2
ubuntu@instance-2:~/lede$ 

似乎是显示的驱动
可能有助于找到错误的编译勾选项：使用测试kernel（5.10），默认编译所有kernel

zhengyiquan, 2021-12-19:
我猜是网络问题？

gili-gili, 2021-12-19:
> 我猜是网络问题？

make clean后在make -j8 download V=s，还是一样

zhengyiquan, 2021-12-19:
打开全局，指定IP

gili-gili, 2021-12-19:
> 打开全局，指定IP

你是指shadowsocks？

zhengyiquan, 2021-12-19:
我之前好像也是你这个问题，后来打开了全局还不行，最后指定IP才行

zhengyiquan, 2021-12-19:
打开了全局还不行，最后指定IP才可以

zhengyiquan, 2021-12-19:
quanju

zhengyiquan, 2021-12-19:
我以前好像也是你这个问题，后来打开了quanju还不行，最后指定IP才行

zhengyiquan, 2021-12-19:
> > 打开，指定IP
> 
> 你是指shadowsocks？

对啊

gili-gili, 2021-12-19:
> 对啊

可我服务器不在国内啊，下载没问题的

closed 2021-12-20 by AmadeusGhost