coolsnowwolf/lede #8486: Inter I225-V 2.5G问题，希望能得到重视。

dengchan0416, 2021-12-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
i225驱动在OP有BUG，经常出现端口突然不能用。四个口只有两口能用，有时一重启之前能用网卡口突然不能用了。爱快暂时没有这个问题，希望能修复一下。

### (2) 路由器型号和固件版本
CPU：J4125  
固件：OpenWrt R21.12.1  内核：5.4.162

### (3) 详细日志

A：

wi11iamzhao, 2021-12-20:
木有日志怎么修呢

helianling, 2021-12-20:
看来还不如螃蟹的8125B 2.5G网卡，这种问题多数是驱动问题，只能等官方驱动更新了。

dengchan0416, 2021-12-21:
> 木有日志怎么修呢
内核日志
[   11.481607] usbcore: registered new interface driver ums-usbat
[   11.493829] xt_time: kernel timezone is -0000
[   11.499689] usbcore: registered new interface driver asix
[   11.505551] usbcore: registered new interface driver ax88179_178a
[   11.529250] PPP generic driver version 2.4.2
[   11.533800] PPP MPPE Compression module registered
[   11.538898] NET: Registered protocol family 24
[   11.558540] usbcore: registered new interface driver rt2800usb
[   11.566181] usbcore: registered new interface driver ath9k_htc
[   11.573274] kmodloader: done loading kernel modules from /etc/modules.d/*
[   15.880939] random: crng init done
[   16.838272] br-lan: port 1(eth0) entered blocking state
[   16.843498] br-lan: port 1(eth0) entered disabled state
[   16.848792] device eth0 entered promiscuous mode
[   16.936324] br-lan: port 2(eth1) entered blocking state
[   16.941541] br-lan: port 2(eth1) entered disabled state
[   16.946835] device eth1 entered promiscuous mode
[   26.365721] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   26.373538] NFSD: Using legacy client tracking operations.
[   26.379005] NFSD: starting 10-second grace period (net f0000097)
[   29.122553] overlayfs: filesystem on '/opt/docker/check-overlayfs-support1515024217/upper' not supported as upperdir
[   29.142885] overlayfs: filesystem on '/opt/docker/check-overlayfs-support3221619096/upper' not supported as upperdir
[   38.756355] igc 0000:02:00.0 eth1: igc: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   38.766893] br-lan: port 2(eth1) entered blocking state
[   38.772130] br-lan: port 2(eth1) entered forwarding state
[   38.777687] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   40.245025] squashfs: Unknown parameter 'discard'
[   40.250456] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
[   40.258015] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   40.265657] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
[   40.273193] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   40.282309] [EXFAT] trying to mount...
[   63.589377] i915 0000:00:02.0: Failed to load DMC firmware i915/glk_dmc_ver1_04.bin. Disabling runtime power management.
[   63.600231] i915 0000:00:02.0: DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[  130.335773] igc 0000:02:00.0 eth1: igc: eth1 NIC Link is Down
[  130.341855] br-lan: port 2(eth1) entered disabled state
[  137.556893] igc 0000:01:00.0 eth0: igc: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[  137.674383] br-lan: port 1(eth0) entered blocking state
[  137.679607] br-lan: port 1(eth0) entered forwarding state

系统日志
Tue Dec 21 08:48:29 2021 daemon.info modprobe: nf_conntrack is already loaded
Tue Dec 21 08:48:29 2021 daemon.info modprobe: nf_conntrack_netlink is already loaded
Tue Dec 21 08:48:32 2021 daemon.info procd: - init complete -
Tue Dec 21 08:48:32 2021 user.notice mwan3[2853]: Using firewall mask 0x3F00
Tue Dec 21 08:48:32 2021 user.notice mwan3[2853]: Max interface count is 60
Tue Dec 21 08:48:32 2021 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Tue Dec 21 08:48:33 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
Tue Dec 21 08:48:39 2021 kern.info kernel: [   38.756355] igc 0000:02:00.0 eth1: igc: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Tue Dec 21 08:48:39 2021 kern.info kernel: [   38.766893] br-lan: port 2(eth1) entered blocking state
Tue Dec 21 08:48:39 2021 kern.info kernel: [   38.772130] br-lan: port 2(eth1) entered forwarding state
Tue Dec 21 08:48:39 2021 daemon.notice netifd: Network device 'eth1' link is up
Tue Dec 21 08:48:39 2021 kern.info kernel: [   38.777687] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Tue Dec 21 08:48:39 2021 daemon.notice netifd: bridge 'br-lan' link is up
Tue Dec 21 08:48:39 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Dec 21 08:48:40 2021 kern.err kernel: [   40.245025] squashfs: Unknown parameter 'discard'
Tue Dec 21 08:48:40 2021 kern.info kernel: [   40.250456] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
Tue Dec 21 08:48:40 2021 kern.err kernel: [   40.258015] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
Tue Dec 21 08:48:40 2021 kern.info kernel: [   40.265657] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
Tue Dec 21 08:48:40 2021 kern.err kernel: [   40.273193] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
Tue Dec 21 08:48:40 2021 kern.info kernel: [   40.282309] [EXFAT] trying to mount...
Tue Dec 21 08:48:48 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Tue Dec 21 08:48:51 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Tue Dec 21 08:49:04 2021 kern.notice kernel: [   63.589377] i915 0000:00:02.0: Failed to load DMC firmware i915/glk_dmc_ver1_04.bin. Disabling runtime power management.
Tue Dec 21 08:49:04 2021 kern.notice kernel: [   63.600231] i915 0000:00:02.0: DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
Tue Dec 21 08:50:10 2021 kern.info kernel: [  130.335773] igc 0000:02:00.0 eth1: igc: eth1 NIC Link is Down
[... cut ...]
Tue Dec 21 08:50:18 2021 kern.info kernel: [  137.556893] igc 0000:01:00.0 eth0: igc: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Tue Dec 21 08:50:18 2021 kern.info kernel: [  137.674383] br-lan: port 1(eth0) entered blocking state
Tue Dec 21 08:50:18 2021 kern.info kernel: [  137.679607] br-lan: port 1(eth0) entered forwarding state
Tue Dec 21 08:50:18 2021 daemon.notice netifd: Network device 'eth0' link is up
Tue Dec 21 08:50:18 2021 daemon.notice netifd: bridge 'br-lan' link is up
Tue Dec 21 08:50:18 2021 daemon.notice netifd: Interface 'lan' has link connectivity

[image]

zhangj3, 2021-12-21:
请使用5.10.X 内核

dengchan0416, 2021-12-22:
> 请使用5.10.X 内核

5.10内核听说不稳定

rockanarki, 2022-01-01:
5.10 也是有同样的问题。只是感觉没影响我个人使用罢了。

CPU：X86
固件：OpenWrt R21.12.1  内核：5.10.88

日志如下:

Fri Dec 31 12:41:02 2021 daemon.notice netifd: Network device 'eth1' link is down
Fri Dec 31 12:41:02 2021 kern.info kernel: [58437.485179] igc 0000:13:00.0 eth1: NIC Link is Down
Fri Dec 31 12:41:02 2021 kern.info kernel: [58437.485482] br-lan: port 2(eth1) entered disabled state
Fri Dec 31 12:41:04 2021 daemon.notice netifd: Network device 'eth1' link is up
Fri Dec 31 12:41:05 2021 kern.info kernel: [58440.469028] igc 0000:13:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Fri Dec 31 12:41:05 2021 kern.info kernel: [58440.469404] br-lan: port 2(eth1) entered blocking state
Fri Dec 31 12:41:05 2021 kern.info kernel: [58440.469555] br-lan: port 2(eth1) entered forwarding state
Fri Dec 31 12:41:21 2021 daemon.notice netifd: Network device 'eth1' link is down
Fri Dec 31 12:41:21 2021 kern.info kernel: [58457.134605] igc 0000:13:00.0 eth1: NIC Link is Down
Fri Dec 31 12:41:21 2021 kern.info kernel: [58457.134974] br-lan: port 2(eth1) entered disabled state
Fri Dec 31 12:41:24 2021 daemon.notice netifd: Network device 'eth1' link is up
Fri Dec 31 12:41:24 2021 kern.info kernel: [58459.908702] igc 0000:13:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Fri Dec 31 12:41:24 2021 kern.info kernel: [58459.910331] br-lan: port 2(eth1) entered blocking state
Fri Dec 31 12:41:24 2021 kern.info kernel: [58459.910604] br-lan: port 2(eth1) entered forwarding state
Fri Dec 31 12:41:35 2021 daemon.notice netifd: Network device 'eth1' link is down
Fri Dec 31 12:41:35 2021 kern.info kernel: [58471.254197] igc 0000:13:00.0 eth1: NIC Link is Down
Fri Dec 31 12:41:35 2021 kern.info kernel: [58471.254459] br-lan: port 2(eth1) entered disabled state
Fri Dec 31 12:41:38 2021 daemon.notice netifd: Network device 'eth1' link is up
Fri Dec 31 12:41:38 2021 kern.info kernel: [58474.288398] igc 0000:13:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Fri Dec 31 12:41:38 2021 kern.info kernel: [58474.288825] br-lan: port 2(eth1) entered blocking state
Fri Dec 31 12:41:38 2021 kern.info kernel: [58474.288992] br-lan: port 2(eth1) entered forwarding state
Fri Dec 31 12:43:55 2021 daemon.err nlbwmon[8449]: Netlink receive failure: Out of memory
Fri Dec 31 12:43:55 2021 daemon.err nlbwmon[8449]: Unable to dump conntrack: No buffer space available
Fri Dec 31 12:46:53 2021 daemon.warn miniupnpd[6842]: upnp_event_process_notify: connect(192.168.3.6:2869): Operation timed out
Fri Dec 31 13:46:38 2021 daemon.err nlbwmon[8449]: Netlink receive failure: Object busy
Fri Dec 31 13:46:38 2021 daemon.err nlbwmon[8449]: Unable to dump conntrack: I/O error
Fri Dec 31 13:48:21 2021 daemon.notice netifd: Network device 'eth1' link is down
Fri Dec 31 13:48:21 2021 kern.info kernel: [62477.341438] igc 0000:13:00.0 eth1: NIC Link is Down
Fri Dec 31 13:48:21 2021 kern.info kernel: [62477.341999] br-lan: port 2(eth1) entered disabled state
Fri Dec 31 13:48:24 2021 daemon.notice netifd: Network device 'eth1' link is up
Fri Dec 31 13:48:24 2021 kern.info kernel: [62479.655663] igc 0000:13:00.0 eth1: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
Fri Dec 31 13:48:24 2021 kern.info kernel: [62479.656232] br-lan: port 2(eth1) entered blocking state
Fri Dec 31 13:48:24 2021 kern.info kernel: [62479.656484] br-lan: port 2(eth1) entered forwarding state
Fri Dec 31 15:28:40 2021 daemon.notice netifd: Network device 'eth1' link is down
Fri Dec 31 15:28:40 2021 kern.info kernel: [68496.068487] igc 0000:13:00.0 eth1: NIC Link is Down
Fri Dec 31 15:28:40 2021 kern.info kernel: [68496.068834] br-lan: port 2(eth1) entered disabled state
Fri Dec 31 15:28:43 2021 daemon.notice netifd: Network device 'eth1' link is up
Fri Dec 31 15:28:43 2021 kern.info kernel: [68499.032710] igc 0000:13:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Fri Dec 31 15:28:43 2021 kern.info kernel: [68499.033277] br-lan: port 2(eth1) entered blocking state
Fri Dec 31 15:28:43 2021 kern.info kernel: [68499.033528] br-lan: port 2(eth1) entered forwarding state
Fri Dec 31 15:29:00 2021 daemon.notice netifd: Network device 'eth1' link is down
Fri Dec 31 15:29:00 2021 kern.info kernel: [68515.637943] igc 0000:13:00.0 eth1: NIC Link is Down
Fri Dec 31 15:29:00 2021 kern.info kernel: [68515.638307] br-lan: port 2(eth1) entered disabled state
Fri Dec 31 15:29:02 2021 kern.info kernel: [68517.922158] igc 0000:13:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Fri Dec 31 15:29:03 2021 daemon.notice netifd: Network device 'eth1' link is up
Fri Dec 31 15:29:03 2021 kern.info kernel: [68518.397055] br-lan: port 2(eth1) entered blocking state
Fri Dec 31 15:29:03 2021 kern.info kernel: [68518.397318] br-lan: port 2(eth1) entered forwarding state
Fri Dec 31 15:29:13 2021 daemon.notice netifd: Network device 'eth1' link is down
Fri Dec 31 15:29:13 2021 kern.info kernel: [68528.917529] igc 0000:13:00.0 eth1: NIC Link is Down
Fri Dec 31 15:29:13 2021 kern.info kernel: [68528.917806] br-lan: port 2(eth1) entered disabled state
Fri Dec 31 15:29:16 2021 daemon.notice netifd: Network device 'eth1' link is up
Fri Dec 31 15:29:16 2021 kern.info kernel: [68531.951775] igc 0000:13:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Fri Dec 31 15:29:16 2021 kern.info kernel: [68531.952167] br-lan: port 2(eth1) entered blocking state
[... cut ...]
Sat Jan  1 15:23:25 2022 kern.info kernel: [154578.414401] br-lan: port 2(eth1) entered disabled state
Sat Jan  1 15:23:28 2022 daemon.notice netifd: Network device 'eth1' link is up
Sat Jan  1 15:23:28 2022 kern.info kernel: [154581.468400] igc 0000:13:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Sat Jan  1 15:23:28 2022 kern.info kernel: [154581.468798] br-lan: port 2(eth1) entered blocking state
Sat Jan  1 15:23:28 2022 kern.info kernel: [154581.468976] br-lan: port 2(eth1) entered forwarding state

rockanarki, 2022-01-02:
后注：重新clone干净编译5.15之后使用未出现上述问题。

shu0802, 2022-05-11:
5.15还是有问题
[image]

coolsnowwolf, 2026-09-05:
I225-V 问题基于 OpenWrt 5.4.162，没有网卡型号批次、PCIe 信息、ethtool 和内核日志；属于旧 igc/硬件问题，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf