coolsnowwolf/lede #8496: luci-app-turboacc缺少依赖

ligehhh, 2021-12-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
luci-app-turboacc缺少依赖
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency dnsproxy for luci-app-turboacc
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-turboacc found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-turboacc-zh-cn:
 * 	dnsproxy
 * opkg_install_cmd: Cannot install package luci-i18n-turboacc-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/lihh/lede'
make[1]: *** [package/Makefile:111: /home/lihh/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lihh/lede'
make: *** [/home/lihh/lede/include/toplevel.mk:230：world] 错误 2
A：
编译luci-app-turboacc缺少依赖
### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

zhiwen1987, 2021-12-21:
同样问题 
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency dnsproxy for luci-app-turboacc
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-turboacc found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-turboacc-zh-cn:
 *      dnsproxy
 * opkg_install_cmd: Cannot install package luci-i18n-turboacc-zh-cn.
package/Makefile:66: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255

hxlls, 2021-12-21:
dnsproxy被移走了，取消turboacc下的turboacc

wi11iamzhao, 2021-12-22:
[dnsproxy被移动了](https://github.com/coolsnowwolf/lede/commit/dbf54e83c3c3cf58594b59316d6217e5dba75f7a)
make distclean之后从头编译即可

closed 2021-12-24 by AmadeusGhost