coolsnowwolf/lede #8501: 正常编译 Lienol 大佬的可道云，刷如固件后运行可道云出现问题，之前php7都是没问题的，php8后有问题了，求大佬帮忙看看

759071793, 2021-12-22:
Fatal error: Cannot acquire reference to $GLOBALS in /mnt/sdc1/kodexplorer/app/function/web.function.php on line 824

breakings, 2021-12-22:
注释掉这个824行就好

759071793, 2021-12-22:
> 注释掉这个824行就好

我尝试注释掉，没有php报错了，但是thinkphp会内部报错，这个是全局变量，注释这招也没办法。

breakings, 2021-12-22:
没办法,可道上游估计都不看php8开发文档的,更别提兼容了

759071793, 2021-12-22:
> 没办法,可道上游估计都不看php8开发文档的,更别提兼容了

好的，感谢大佬回复

759071793, 2021-12-22:
> 

另外在请教下，最新lede源码是不带php7了吗，默认只有php8了，menuconfig里面找不到php7的选项了

breakings, 2021-12-22:
> > 
> 
> 另外在请教下，最新lede源码是不带php7了吗，默认只有php8了，menuconfig里面找不到php7的选项了

没有了,有需求可以svn openwrt 官方21.02分支

759071793, 2021-12-22:
> > > 
> > 
> > 
> > 另外在请教下，最新lede源码是不带php7了吗，默认只有php8了，menuconfig里面找不到php7的选项了
> 
> 没有了,有需求可以svn openwrt 官方21.02分支

收到，谢谢，打扰了

759071793, 2021-12-22:
此帖终结，kodbox官方貌似还不支持php8.1

closed 2021-12-22 by 759071793

WYC-2020, 2022-01-23:
https://github.com/coolsnowwolf/lede/issues/8636#issuecomment-1019453416 @759071793 @breakings @zx2c4 @ynezz @jesset