coolsnowwolf/lede #8503: 启用FullCone-NAT的选项，勾选后生效，但是重启后就不生效了

cn-404-not-found, 2021-12-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ]有类似的issue，详见https://github.com/coolsnowwolf/lede/issues/7858

## 2. 详细叙述

### (1) 具体问题

A：在防火墙位置开启启用FullCone-NAT后，可以在Turbo ACC中看到已经生效，实测也生效了，但是重启路由器后，防火墙里面看到是未勾选“启用FullCone-NAT”的状态，需要重新开启

### (2) 路由器型号和固件版本

A：斐讯N1

### (3) 详细日志

A：无

jxph1123, 2021-12-23:
在Turbo ACC中勾选全锥型NAT，重启后还在，防火墙也在

cn-404-not-found, 2021-12-23:
> 在Turbo ACC中勾选全锥型NAT，重启后还在，防火墙也在

在Turbo ACC根本就没有让你选择的地方，BBR都不知道怎么开启

[image]

sdf8057, 2021-12-23:
斐讯n1只能自食其力

jxph1123, 2021-12-23:
> > 在Turbo ACC中勾选全锥型NAT，重启后还在，防火墙也在
> 
> 在Turbo ACC根本就没有让你选择的地方，BBR都不知道怎么开启
> 
> [image]

你这个界面都不对，从新编译吧

renfei, 2021-12-26:
> > 在Turbo ACC中勾选全锥型NAT，重启后还在，防火墙也在
> 
> 在Turbo ACC根本就没有让你选择的地方，BBR都不知道怎么开启
> 
> [image]

界面跟我长得不一样呢，我的有设置呀：

[image]

coolsnowwolf, 2026-09-06:
已核对当前 luci-app-turboacc：配置持久化保存 turboacc.config.fullcone，服务启动时会读取该值并同步 firewall.@defaults[0].fullcone 后重载防火墙；首次启动脚本也会初始化该配置。原 issue 中重启后状态丢失的问题已由现行实现覆盖，关闭为 completed。

closed 2026-09-06 by coolsnowwolf