coolsnowwolf/lede #8504: 编译失败

flying864, 2021-12-22:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
编译失败，网络没有问题，
make -j8 download V=s
后库全部下载完成，但是重试好几次还是同样的问题！

### (3) 详细日志
make[2]: Leaving directory '/home/fine/openwrt'
make[1]: *** [package/Makefile:110: /home/fine/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fine/openwrt'
make: *** [/home/fine/openwrt/include/toplevel.mk:230：world] 错误 2

xiangfeigao, 2021-12-22:
我也是同样问题

kevinCSDN, 2021-12-23:
切换个节点再make -j12 V=s（12核心）

zhoucheng45, 2021-12-23:
我也是编译失败。

Building Go toolchain2 using go_bootstrap and Go toolchain1.
Building Go toolchain3 using go_bootstrap and Go toolchain2.
Building packages and commands for linux/amd64.
touch /home/benny/lede/build_dir/hostpkg/go-1.17.3/.built
rm -rf "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/openwrt"
rm -f "/home/benny/lede/staging_dir/hostpkg/bin/go"
rm -f "/home/benny/lede/staging_dir/hostpkg/bin/gofmt"
rm -rf "/home/benny/lede/staging_dir/hostpkg/lib/go-cross"
rm -rf "/home/benny/lede/staging_dir/hostpkg/share/go-cross"
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/lib/go-cross"
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/share/go-cross"
cp -fpR "/home/benny/lede/build_dir/hostpkg/go-1.17.3/api" "/home/benny/lede/staging_dir/hostpkg/share/go-cross/"
ln -sf "../../share/go-cross/api" "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/"
install -m0644 -p "/home/benny/lede/build_dir/hostpkg/go-1.17.3/VERSION" "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/"
for file in AUTHORS CONTRIBUTING.md CONTRIBUTORS LICENSE PATENTS README.md SECURITY.md; do if [ -f "/home/benny/lede/build_dir/hostpkg/go-1.17.3/$file" ]; then install -m0644 -p "/home/benny/lede/build_dir/hostpkg/go-1.17.3/$file" "/home/benny/lede/staging_dir/hostpkg/share/go-cross/" ; fi ; done
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/bin"
install -m0755 -p "/home/benny/lede/build_dir/hostpkg/go-1.17.3/bin"/* "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/bin/"
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/pkg"
cp -fpR "/home/benny/lede/build_dir/hostpkg/go-1.17.3/pkg/linux_amd64" "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/pkg/"
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/pkg/tool/linux_amd64"
install -m0755 -p "/home/benny/lede/build_dir/hostpkg/go-1.17.3/pkg/tool/linux_amd64"/* "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/pkg/tool/linux_amd64/"
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/lib/go-cross"
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/share/go-cross"
cp -fpR "/home/benny/lede/build_dir/hostpkg/go-1.17.3/lib" "/home/benny/lede/staging_dir/hostpkg/share/go-cross/"
ln -sf "../../share/go-cross/lib" "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/"
cp -fpR "/home/benny/lede/build_dir/hostpkg/go-1.17.3/misc" "/home/benny/lede/staging_dir/hostpkg/share/go-cross/"
ln -sf "../../share/go-cross/misc" "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/"
cp -fpR "/home/benny/lede/build_dir/hostpkg/go-1.17.3/src" "/home/benny/lede/staging_dir/hostpkg/share/go-cross/"
ln -sf "../../share/go-cross/src" "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/"
cp -fpR "/home/benny/lede/build_dir/hostpkg/go-1.17.3/test" "/home/benny/lede/staging_dir/hostpkg/share/go-cross/"
ln -sf "../../share/go-cross/test" "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/"
find "/home/benny/lede/staging_dir/hostpkg/share/go-cross/src/" \! -type d -a \( -name "*.bat" -o -name "*.rc" \) -delete
if [ -d "/home/benny/lede/build_dir/hostpkg/go-1.17.3/pkg/include" ]; then install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/pkg" ; install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/share/go-cross/pkg" ; cp -fpR "/home/benny/lede/build_dir/hostpkg/go-1.17.3/pkg/include" "/home/benny/lede/staging_dir/hostpkg/share/go-cross/pkg/" ; ln -sf "../../../share/go-cross/pkg/include" "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/pkg/" ; fi
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/bin"
ln -sf "../lib/go-cross/bin/go" "/home/benny/lede/staging_dir/hostpkg/bin/go"
ln -sf "../lib/go-cross/bin/gofmt" "/home/benny/lede/staging_dir/hostpkg/bin/gofmt"
rm -rf "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/pkg/linux_amd64"
install -d -m0755 "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/openwrt"
install -m0755 ./files/go-gcc-helper "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/openwrt/"
ln -sf go-gcc-helper "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/openwrt/gcc"
ln -sf go-gcc-helper "/home/benny/lede/staging_dir/hostpkg/lib/go-cross/openwrt/g++"
mkdir -p /home/benny/lede/staging_dir/hostpkg/stamp
touch /home/benny/lede/build_dir/hostpkg/go-1.17.3/.built
touch /home/benny/lede/staging_dir/hostpkg/stamp/.golang_installed
make[3]: Leaving directory '/home/benny/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#240.13#63.04#213.65
make[2]: Leaving directory '/home/benny/lede'
make[1]: *** [package/Makefile:110: /home/benny/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/benny/lede'
make: *** [/home/benny/lede/include/toplevel.mk:230：world] 错误 2

zhoucheng45, 2021-12-23:
> 切换个节点再make -j12 V=s（12核心）

切换节点，这儿的节点指的市啥？小白求指教

zhoufeiyu2001, 2021-12-25:
我也是一样的错误

zhoucheng45, 2021-12-25:
您好，您的邮件我已经收到！请放心！，我会尽快回复！若有急事，请拨打15927970512！，祝工作顺利，身体健康！

alexliu269, 2021-12-25:
我这里也这样。一样的报错

sbwcwfhv001, 2021-12-25:
make[3]: *** [Makefile:160: /home/mybook/Desktop/lede/build_dir/target-x86_64_musl/dockerd-20.10.12/.built] Error 1
make[3]: Leaving directory '/home/mybook/Desktop/lede/feeds/packages/utils/dockerd'
time: package/feeds/packages/dockerd/compile#285.24#69.54#224.67
    ERROR: package/feeds/packages/dockerd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/dockerd/compile] Error 1
make[2]: Leaving directory '/home/mybook/Desktop/lede'
make[1]: *** [package/Makefile:110: /home/mybook/Desktop/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mybook/Desktop/lede'
make: *** [/home/mybook/Desktop/lede/include/toplevel.mk:230：world] 错误 2

zhoufeiyu2001, 2021-12-26:
最新版本我在执行make clean后重新编译已经可以成功了

sbwcwfhv001, 2021-12-27:
> 最新版本我在执行make clean后重新编译已经可以成功了

我还是不行不知道是什么问题

closed 2021-12-29 by AmadeusGhost