coolsnowwolf/lede #8514: ath11k驱动在线编译报错ath11k.ko' is missing

Dimitri2020007, 2021-12-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

2021-12-23T16:50:07.8824683Z ERROR: module '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath11k/ath11k.ko' is missing.
2021-12-23T16:50:07.8826927Z make[3]: *** [Makefile:574: /workdir/openwrt/bin/targets/x86/64/packages/kmod-ath11k_5.10.87+5.10.42-1-3_x86_64.ipk] Error 1
2021-12-23T16:50:07.8827931Z make[3]: Leaving directory '/workdir/openwrt/package/kernel/mac80211'
2021-12-23T16:50:07.8830961Z time: package/kernel/mac80211/compile#0.40#0.09#0.44
2021-12-23T16:50:07.8838840Z     ERROR: package/kernel/mac80211 failed to build.
2021-12-23T16:50:07.8847857Z make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
2021-12-23T16:50:07.8860579Z make[2]: Leaving directory '/workdir/openwrt'
2021-12-23T16:50:07.8880341Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-12-23T16:50:07.8881385Z make[1]: Leaving directory '/workdir/openwrt'
2021-12-23T16:50:07.8882459Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2021-12-23T16:50:07.9010593Z ##[error]Process completed with exit code 2.

paldier, 2021-12-25:
你一个x86编译他干嘛，那个700多的网卡也不能用这个跑

sonataaaa, 2022-01-13:
https://libredd.it/r/homelab/comments/mwkbed/80211ax_ath11kqcn9074_homebrew_router_running/
这有个linux 跑hostapd 的，openwrt不行的话先这样对付吧

closed 2022-01-24 by AmadeusGhost