coolsnowwolf/lede #8520: 咨询个多线多拨的问题，我认为它不应该成功才对，求大佬帮忙解答，有固件bug无关

zhouxiaolong920, 2021-12-24:
有openwrt bug无关，只是求大佬，解答一下：我也算半个网络工程师，比较熟悉h3c,huawei各种网络设备的配置，基础的网络知识没有问题
----
背景：很多光猫 多lan口，接到openwrt多wlan口上，通过多次pppoe拨号，然后设置负载，测速实现了1200M左右带宽，我很不理解求解，基于这个背景，求针对性的回答一下：

问题： 多个pppoe拨号，理论上回造成，回包不一致的问题（互联网访问openwrt内部服务的情况）
问题描述：当内部服务 映射到外网的时候，互联上通过外网ip访问到内部应用，会形成五元组 src_ip,src_port,protocal,dst_ip,dst_port，当openwrt收到包之后，返回数据包的时候，回包就是有问题，因为默认网关有多个地址，我不知道openwrt是否有问题，因为华为这类企业防火墙会这样的问题，为了解决这个问题，必须强制指定vlan或内网ip走某个默认网关出去（本地路由策略或流分类，流行为，流策略可以解决这个问题），这也导致了单ip是没速无法突破1000M,多个ip，是可以突破1000M。

zhouxiaolong920, 2021-12-24:
求个权威一点的解答。

lexin8, 2021-12-26:
多拨谁玩openwrt，支持多拨地区越来越少，再加上openwrt流策略难用，开发在更少。我多拨用了一段时间就不用了，问题很多。

minlang112, 2021-12-26:
> 多拨谁玩openwrt，支持多拨地区越来越少，再加上openwrt流策略难用，开发在更少。我多拨用了一段时间就不用了，问题很多。

不止是openwrt，ROS一样难用，现在都转https了，多拨也就P2P时候有点效果。

coolsnowwolf, 2026-09-06:
报告正文已明确与 OpenWrt/LEDE bug 无关，属于多线多拨回程路由的原理咨询，不涉及仓库代码缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf