coolsnowwolf/lede #8530: openwrt做主路由，外网无法通过openwrt端口转发连接内网的l2tp/ipsec vpn服务器

bonnyandsky, 2021-12-27:
搜索了各种 都没有类似案例
具体情况如下：
j4125的x86平台openwrt做主路由器，内网有一台群晖部署了l2tp/ipsec的vpn服务
openwrt已经做了端口转发和防火墙端口放行设置。见下图：
[image]
[image]
使用手机作为客户端，在局域网连接openwrt拨号的公网ip，可以连上群晖的vpn服务。具体如下 ：
openwrt拨号得到公网ip：116.237.253.101
[image]
安卓手机设置vpn服务器地址ip：116.237.253.101
[image]
手机在局域网内，连接群晖vpn服务器成功
[image]
手机使用5g网络，在外网连接群晖服务器失败
[image]
因为在19年发生过设置了端口转发，实际会有部分端口转发失败的经历。当时在防火墙--自定义规则--内设置下面内容：
iptables -t nat -I PREROUTING -i pppoe-wan -p udp --dport 1701 -j DNAT --to-destination 192.168.0.100:1701
iptables -t nat -I PREROUTING -i pppoe-wan -p udp --dport 500 -j DNAT --to-destination 192.168.0.100:500
iptables -t nat -I PREROUTING -i pppoe-wan -p udp --dport 4500 -j DNAT --to-destination 192.168.0.100:4500
重启op后是可以正常连接群晖上的vpn服务器的。
但是这次设置后，问题依然无法解决。

Wilson-coder, 2021-12-28:
我用的是Zerotier，没有配置L2TP/IPSec。你自己看一下这个帖子：https://serverfault.com/questions/451381/which-ports-for-ipsec-lt2p

bonnyandsky, 2021-12-28:
> 我用的是Zerotier，没有配置L2TP/IPSec。你自己看一下这个帖子：https://serverfault.com/questions/451381/which-ports-for-ipsec-lt2p

谢谢，之所以用l2tp是因为windows自带客户端。这样我就不需要额外安装软件了。在任何一台电脑上设置下就可以连接到家里的vpn使用了

bonnyandsky, 2021-12-29:
经过大佬指点，问题解决了。是加速的问题，SFE关闭就可以连上了。

closed 2021-12-29 by bonnyandsky