coolsnowwolf/lede #8531: 求助，无法跟一些ip正常建立TCP连接

Weiers, 2021-12-27:
在一台主机上建立连接抓包如下：
No.	Time	Source	Destination	Protocol	Length	Info
294	2.980172	192.168.123.99	106.12.125.240	TCP	74	58454 → 8655 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=3458915874 TSecr=0 WS=128
295	2.980219	192.168.123.1	192.168.123.99	TCP	74	58454 → 8655 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=3458915874 TSecr=0 WS=128
297	2.980339	192.168.123.99	192.168.123.1	TCP	54	8655 → 58454 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
298	2.980361	106.12.125.240	192.168.123.99	TCP	54	8655 → 58454 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0

主机ip 123.99,路由ip 123.1，目标ip 106.12.125.240，主机向目标发送SYN，不知为何路由发了一条一样的SYN给主机，接着就是两个Reset的返回。

在路由下同样与目标建立连接就没有问题，抓包如下：
No.	Time	Source	Destination	Protocol	Length	Info
209	3.156376	125.38.*.*	106.12.125.240	TCP	82	50200 → 8655 [SYN] Seq=0 Win=64980 Len=0 MSS=1444 SACK_PERM=1 TSval=1788754448 TSecr=0 WS=128
213	3.194153	106.12.125.240	125.38.*.*	TCP	74	8655 → 50200 [SYN, ACK] Seq=0 Ack=1 Win=28650 Len=0 MSS=1352 SACK_PERM=1 WS=128
214	3.194188	125.38.*.*	106.12.125.240	TCP	62	50200 → 8655 [ACK] Seq=1 Ack=1 Win=65024 Len=0
215	3.194498	125.38.*.*	106.12.125.240	TLSv1.2	420	Client Hello

其中125.38.*.*是路由的公网ip。
跪求哪位大神给点儿思路，是防火墙设置的问题么？

PyXiaoxin, 2021-12-30:
NAT是不是设置了啥？可能是NAT后的包出了异常导致服务器直接返回RST了

Weiers, 2021-12-30:
> NAT是不是设置了啥？可能是NAT后的包出了异常导致服务器直接返回RST了

恩，找到问题了，是我防火墙设置了那个段的端口映射，然后映射的时候把从内网来的包也给映射了~哈哈~谢谢了~结贴了~

closed 2021-12-30 by Weiers