coolsnowwolf/lede #8535: CIFSD/KSMBD共享新建和编辑EXCEL报错

wozualong, 2021-12-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
luci-app-cifsd挂载共享后，新建EXCEL和编辑EXCEL文件均报错，电脑文件浏览器会崩溃闪退，查询了KSMBD源的ISSUES后，发现作者貌似在3.4.3版本中已经修复，能力实在有限，自行编译报错。。。希望大佬能更新修复下
https://github.com/cifsd-team/ksmbd/issues/545

### (2) 路由器型号和固件版本

A：
K2P,竞斗云,newifi，OpenWrt R21.12.1 / LuCI Master (git-21.335.48743-5f363d9)

### (3) 详细日志

A：
[42989.955757] ksmbd: parse_sec_desc:814: parse_sec_desc: Error -22 mapping Owner SID to uid
[42989.972079] ksmbd: parse_sec_desc:829: parse_sec_desc: Error -22 mapping Group SID to gid
[42989.988388] ksmbd: parse_dacl:479: parse_dacl: Error -22 mapping Owner SID to uid
[42990.003306] ksmbd: parse_dacl:479: parse_dacl: Error -22 mapping Owner SID to uid

WYC-2020, 2021-12-28:
共享？samba4 不香吗？

wozualong, 2021-12-28:
> 共享？samba4 不香吗？

SAMBA4大文件传输会计算剩余时间超时然后失败。。。。不知道怎么解决

WYC-2020, 2021-12-28:
@wozualong 还行吧，我这里复制6个G 没出息你说的那个问题呢，等会我看下这个库吧

wozualong, 2021-12-28:
> 吧，我这里复制6个G 没出息你说的那个问题呢，等会我看下这个库吧

可能我用的是USB2.0吧，用SAMBA4挂载，复制到路由器的U盘速度只有十几兆，400M大小计算时间就超时拷贝失败了。用samba3么，拷贝1个G的大小貌似没问题了，但竞斗云挂载后出现重启LAN口不能自动DHCP，只能插拔网线后或者等三四分钟后才能正常识别。KSMBD共享么就是这个EXCEL有点问题哈

WYC-2020, 2021-12-28:
@wozualong 试下这个吧https://github.com/WYC-2020/packages/commit/b787fbc61f1c6a28e83fe77d931b59b337c683bc 
我这里升级了，有一个日志我去掉了目前没找到解决方法 ，功能不影响，你试下这个看下解决了没

wozualong, 2021-12-28:
> @wozualong 试下这个吧[WYC-2020/packages@b787fbc](https://github.com/WYC-2020/packages/commit/b787fbc61f1c6a28e83fe77d931b59b337c683bc) 我这里升级了，有一个日志我去掉了目前没找到解决方法 ，功能不影响，你试下这个看下解决了没

感谢大佬，编译测试了下，跟excel格式兼容性还是有问题，新建第一个excel文档可以，重复新建多个excel文件会卡住一段较长时间后创建成功，编辑保存还是不行，失去响应，看来这个ksmbd内核更新还是没有彻底解决这问题

WYC-2020, 2021-12-28:
@wozualong 明天我在升级到最新的版本你再看下吧

WYC-2020, 2021-12-29:
把这个也更新一下看下https://github.com/WYC-2020/packages/commit/fbeb0035b2ccdcac81a1b559ca61bd4cdc1c509d，我这边测试没你说的哪个问题，excel我建立了多个，并且直接修改保存都没问题，如果还不行可能你的看下你的操作或者环境了，我这边用x86测试的 @wozualong

WYC-2020, 2021-12-29:
@wozualong 我这边编译x86 测试没你哪个问题

wozualong, 2021-12-29:
> @wozualong 我这边编译x86 测试没你哪个问题

嗯，这个文件昨天也自己改了，今天刚编译了份K2P硬改的，确实没问题了，昨天测试的是竞斗云测试出现的问题，晚点再重新编译个竞斗云的再测下看看，大概率是我自己编译的问题了

coolsnowwolf, 2026-09-06:
已核对当前 packages：ksmbd-tools 已更新到 3.5.2，明显高于 issue 提到已修复问题的 3.4.3，当前 ksmbd 配置和服务脚本也已采用新版路径。原报告针对 R21.12.1 的旧版本问题已被当前包版本覆盖，关闭为 completed。

closed 2026-09-06 by coolsnowwolf