coolsnowwolf/lede #8537: 最近的内核 是滞修复了 ESXI下igb驱动的问题?    一直编译 5.4.140 很难受啊 要改好多东西!!!!

LaierX, 2021-12-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近的内核 是否修复了 ESXI下igb驱动的问题? 
     一直编译 5.4.140 很难受啊 要改好多东西!!!!

### (2) 路由器型号和固件版本

A：ESXI x86 igb网卡

### (3) 详细日志

A：igb网卡一直重置

WYC-2020, 2021-12-29:
这个拿140 和最新的对比那里有区别 做一个补丁不就可以了

LaierX, 2021-12-29:
> 这个拿140 和最新的对比那里有区别 做一个补丁不就可以了

应该是内核内部的相关问题

不是openwrt的

我在论坛上看的
不知对不对

LaierX, 2021-12-29:
替换 内核 的驱动 解决了.. 
用的intel 5.8.5 最新版本的..
下载地址
https://www.intel.com/content/www/us/en/download/14098/intel-network-adapter-driver-for-82575-6-82580-i350-and-i210-211-based-gigabit-network-connections-for-linux.html?wapkw=igb

closed 2021-12-29 by LaierX

yuhua18, 2022-01-20:
> 替换 内核 的驱动 解决了.. 用的intel 5.8.5 最新版本的.. 下载地址 https://www.intel.com/content/www/us/en/download/14098/intel-network-adapter-driver-for-82575-6-82580-i350-and-i210-211-based-gigabit-network-connections-for-linux.html?wapkw=igb

怎么替换

WYC-2020, 2022-01-20:
@LaierX 我仓库5.4和5.10默认编译出来就是5.8.5，5.15内核编译是自带的，因为5.15内核变化太大，5.8.5不适合

dudulangjiao, 2022-01-26:
> 替换 内核 的驱动 解决了.. 用的intel 5.8.5 最新版本的.. 下载地址 https://www.intel.com/content/www/us/en/download/14098/intel-network-adapter-driver-for-82575-6-82580-i350-and-i210-211-based-gigabit-network-connections-for-linux.html?wapkw=igb

麻烦问一下怎么替换？给一个链接教程也行，谢谢了！

WYC-2020, 2022-02-19:
https://github.com/WYC-2020/lede/commit/a232fda3786cf194d311c0477a3f02d79a8d7b68
https://github.com/WYC-2020/lede/commit/d2849c4a2ef2dce3802bc4be91399e95fef308c4
这是添加方法，但是今天我突然想刷到我的软路由上面，发现这个驱动不行，所以我还原了，如果你确认这个驱动你那边可以 你就这样改就行了
@dudulangjiao

dudulangjiao, 2022-02-19:
已转PVE了，哎

ericshunhawk, 2023-01-17:
> 

> https://github.com/WYC-2020/lede/commit/a232fda3786cf194d311c0477a3f02d79a8d7b68 https://github.com/WYC-2020/lede/commit/d2849c4a2ef2dce3802bc4be91399e95fef308c4 这是添加方法，但是今天我突然想刷到我的软路由上面，发现这个驱动不行，所以我还原了，如果你确认这个驱动你那边可以 你就这样改就行了 @dudulangjiao

看不到教程了 请问你那还有替换驱动教程么