coolsnowwolf/lede #8539: 5.10.83之后的版本无法上网

kleisth, 2021-12-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5.10.83之后的版本不知道更新了什么，保留配置更新无法上网。清除配置后重新设置拨号和openclash都可以正常上网和科学上网，一旦到dhcp/dns那里设置静态dhcp，保存后，就又上不了网了。
出问题时，路由本身可以解析域名，ping外网ip也都能通，但客户机ping外网ip都无法ping通，但域名能正常解析
实在不知道是哪的问题了，只要一动dhcp/dns那里的设置，客户机就全上不了了。

### (2) 路由器型号和固件版本

A：x86  5.10.88

### (3) 详细日志

A：日志详细对比了5.10.83和5.10.88的，开机后没什么差别，可以说是一模一样。

kleisth, 2021-12-28:
以防万一，还是把日志也贴上来
`Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.282469] pci 0000:01:00.3: reg 0x10: [mem 0xfe200000-0xfe27ffff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.282478] pci 0000:01:00.3: reg 0x1c: [mem 0xfe28c000-0xfe28ffff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.283540] pci 0000:01:00.3: PME# supported from D0 D3hot D3cold
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.284425] pci 0000:00:1c.0: PCI bridge to [bus 01]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.290697] pci 0000:00:1c.0:   bridge window [mem 0xfe000000-0xfe3fffff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.292464] pci 0000:00:1c.0:   bridge window [mem 0xfce00000-0xfcffffff 64bit pref]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.292464] pci 0000:00:1c.1: PCI bridge to [bus 02]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.294160] pci 0000:00:1c.1:   bridge window [mem 0xfe800000-0xfe9fffff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.294160] pci 0000:00:1c.1:   bridge window [mem 0xfcc00000-0xfcdfffff 64bit pref]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.294160] pci 0000:00:1c.2: PCI bridge to [bus 03]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.295202] pci 0000:00:1c.2:   bridge window [mem 0xfe600000-0xfe7fffff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.295202] pci 0000:00:1c.2:   bridge window [mem 0xfca00000-0xfcbfffff 64bit pref]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.295203] pci 0000:00:1c.3: PCI bridge to [bus 04]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.297856] pci 0000:00:1c.3:   bridge window [mem 0xfe400000-0xfe5fffff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.297856] pci 0000:00:1c.3:   bridge window [mem 0xfc800000-0xfc9fffff 64bit pref]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.297856] pci_bus 0000:05: extended config space not accessible
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.302471] pci 0000:05:01.0: [1b36:0001] type 01 class 0x060400
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.302735] pci 0000:05:01.0: reg 0x10: [mem 0xfde00000-0xfde000ff 64bit]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.304362] pci 0000:05:02.0: [1b36:0001] type 01 class 0x060400
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.306013] pci 0000:05:02.0: reg 0x10: [mem 0xfde01000-0xfde010ff 64bit]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.307650] pci 0000:05:03.0: [1b36:0001] type 01 class 0x060400
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.309286] pci 0000:05:03.0: reg 0x10: [mem 0xfde02000-0xfde020ff 64bit]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.312464] pci 0000:05:04.0: [1b36:0001] type 01 class 0x060400
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.312464] pci 0000:05:04.0: reg 0x10: [mem 0xfde03000-0xfde030ff 64bit]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.315372] pci 0000:00:1e.0: PCI bridge to [bus 05-09] (subtractive decode)
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.315372] pci 0000:00:1e.0:   bridge window [io  0x1000-0x4fff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.315372] pci 0000:00:1e.0:   bridge window [mem 0xfd600000-0xfdffffff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.315372] pci 0000:00:1e.0:   bridge window [mem 0xfc000000-0xfc7fffff 64bit pref]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.315372] pci 0000:00:1e.0:   bridge window [io  0x0000-0x0cf7 window] (subtractive decode)
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.315372] pci 0000:00:1e.0:   bridge window [io  0x0d00-0xffff window] (subtractive decode)
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.315372] pci 0000:00:1e.0:   bridge window [mem 0x000a0000-0x000bffff window] (subtractive decode)
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.322470] pci 0000:00:1e.0:   bridge window [mem 0x40000000-0xafffffff window] (subtractive decode)
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.322470] pci 0000:00:1e.0:   bridge window [mem 0xc0000000-0xfebfffff window] (subtractive decode)
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.322470] pci 0000:00:1e.0:   bridge window [mem 0x100000000-0x8ffffffff window] (subtractive decode)
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.322470] pci_bus 0000:06: extended config space not accessible
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.326946] pci 0000:06:03.0: [1af4:1002] type 00 class 0x00ff00
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.326946] pci 0000:06:03.0: reg 0x10: [io  0x4000-0x403f]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.326946] pci 0000:06:03.0: reg 0x20: [mem 0xfc600000-0xfc603fff 64bit pref]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.330464] pci 0000:06:07.0: [8086:2922] type 00 class 0x010601
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.332464] pci 0000:06:07.0: reg 0x20: [io  0x4040-0x405f]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.332464] pci 0000:06:07.0: reg 0x24: [mem 0xfdc00000-0xfdc00fff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.339575] pci 0000:05:01.0: PCI bridge to [bus 06]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.339575] pci 0000:05:01.0:   bridge window [io  0x4000-0x4fff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.339575] pci 0000:05:01.0:   bridge window [mem 0xfdc00000-0xfddfffff]
Wed Dec 29 01:44:44 2021 kern.info kernel: [    0.339575] pci 0000:05:01.0:   bridge window [mem 0xfc600000-0xfc7fffff 64bit pref]
[... cut ...]
Wed Dec 29 01:45:03 2021 daemon.info avahi-daemon[3888]: New relevant interface ztgvwq4ol3.IPv6 for mDNS.
Wed Dec 29 01:45:03 2021 daemon.info avahi-daemon[3888]: Registering new address record for fe80::b03e:ddff:fe4e:2c0d on ztgvwq4ol3.*.
Wed Dec 29 01:45:03 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Dec 29 01:45:03 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Dec 29 01:45:03 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain dsm.kleist.cn
Wed Dec 29 01:45:04 2021 daemon.info procd: - init complete -
`

dfbucy090, 2021-12-29:
请问如何进入编译5.10内核，按教程步骤只能编译出5.4内核，需要更改哪里或从哪里进入？我不是云编译，请给出具体步骤和命令，本人纯小白，请指教，不胜感激。

kleisth, 2021-12-29:
> 

最新的源码默认就是5.10.x了

d975xbx, 2021-12-29:
> 5.10 内核，其他人只能自己编译出来 5.4 内核，需要改变或从哪里进入？

lede/target/linux/x86/Makefile

suizhenmin, 2021-12-29:
重新同步，现在已经切换到5.10 内核了

dfbucy090, 2021-12-29:
> > 5.10 内核，其他人只能自己编译出来 5.4 内核，需要改变或从哪里进入？
> 
> lede/target/linux/x86/Makefile

谢谢您！

closed 2021-12-31 by coolsnowwolf