coolsnowwolf/lede #8541: x86软路由报 Netlink receive failure: Out of memory

liuyanantj, 2021-12-30:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：系统日志隔一个多小时就报下面的错误
Thu Dec 30 09:44:02 2021 daemon.err nlbwmon[3651]: Netlink receive failure: Out of memory
Thu Dec 30 09:44:02 2021 daemon.err nlbwmon[3651]: Unable to dump conntrack: No buffer space available

### (2) 路由器型号和固件版本
主机名 | OpenWrt
主机型号 | Intel(R) Celeron(R) J4125 CPU @ 2.00GHz : 4 Core 4 Thread 
固件版本 | OpenWrt R21.12.1 / LuCI Master (git-21.335.48743-5f363d9)
内核版本 | 5.10.88
网卡8125
### (3) 详细日志
每隔一个多小时就报这个错误，不知道为什么
Thu Dec 30 16:34:54 2021 daemon.err nlbwmon[3651]: Netlink receive failure: Out of memory
Thu Dec 30 16:34:54 2021 daemon.err nlbwmon[3651]: Unable to dump conntrack: No buffer space available

moonerick, 2021-12-30:
我也遇到了这个问题，解决方案是把宽带监控的提交时间由4H改为2H

liuyanantj, 2021-12-31:
> 我也遇到了这个问题，解决方案是把宽带监控的提交时间由4H改为2H 另外我打算在下一次编译更新时取消这个宽带监控功能，感觉没什么用处

我一直就没把带宽监控那个编译进去  昨天晚上我又重新编译了代码  看到Turbo ACC这个组件更换了  然后晚上到现在 没见到报错误了

realconnor, 2021-12-31:
https://github.com/coolsnowwolf/lede/issues/7965#issuecomment-931843652

coolsnowwolf, 2026-09-06:
nlbwmon conntrack netlink 缓冲不足与连接跟踪表/系统资源有关，缺少表大小、内存和当前版本对照，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf