coolsnowwolf/lede #8543: kmod-fs-cifs 缺少 libdes.ko 依赖

CuteTony, 2021-12-30:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：kmod-fs-cifs 缺失依赖 libdes.ko

### (2) 路由器型号和固件版本

A：/

### (3) 详细日志

A：SHELL= flock /home/tony/lede/tmp/.root-copy.flock -c 'cp -fpR /home/tony/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/packages/.pkgdir/kernel/. /home/tony/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/'
touch /home/tony/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.kernel_installed
Package kmod-fs-cifs is missing dependencies for the following libraries:
libdes.ko
make[3]: *** [modules/fs.mk:120: /home/tony/lede/bin/targets/armvirt/64/packages/kmod-fs-cifs_5.4.162-1_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/home/tony/lede/package/kernel/linux'
time: package/kernel/linux/compile#1.95#0.16#2.10
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/tony/lede'
make[1]: *** [package/Makefile:110: /home/tony/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tony/lede'
make: *** [/home/tony/lede/include/toplevel.mk:230: world] Error 2

CuteTony, 2021-12-30:
可能与 commit [fc0f5e3](https://github.com/coolsnowwolf/lede/commit/fc0f5e35b1274d73056418b90d4a7c2798ecaa7f) 中移除了 kmod-crypto-des 有关。

coolsnowwolf, 2021-12-30:
已修复

CuteTony, 2021-12-30:
> 已修复

感谢 L 大。拉取更新后编译通过，现 close issue.

closed 2021-12-30 by CuteTony