coolsnowwolf/lede #8546: kmod_mmc_spi.so missing

GreatMichaelLee, 2021-12-30:
请问怎么解决？

echo "bnx2-firmware" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/builder/lede_x86/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#0.47#2.21#1.99
make[3]: Entering directory '/home/builder/lede_x86/package/firmware/prism54-firmware'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/builder/lede_x86/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.20#0.08#0.25
make[3]: Entering directory '/home/builder/lede_x86/package/kernel/linux'
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-mmc-spi is missing dependencies for the following libraries:
of_mmc_spi.ko
make[3]: *** [modules/spi.mk:27: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-mmc-spi_5.15.10-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#2.65#0.24#2.86
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

GreatMichaelLee, 2021-12-30:
把这个mmc-spi的从Kernel library里去掉以后，又报其他错了，什么鬼

rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-codec-hdmi/lib/modules/5.15.10/snd-hda-codec-hdmi.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-codec-hdmi into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-sound-hda-codec-hdmi_5.15.10-1_x86_64.ipk
removed '/home/builder/lede_x86/bin/targets/x86/64/packages/kmod-sound-hda-intel_5.15.10-1_x86_64.ipk'
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-intel/lib/modules/5.15.10/snd-intel-dspcfg.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-intel/lib/modules/5.15.10/snd-hda-intel.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-intel into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-sound-hda-intel_5.15.10-1_x86_64.ipk
removed '/home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-core_5.10.88-1_x86_64.ipk'
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.10/drivers/usb/core/usbcore.ko' is built-in.
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.10/drivers/usb/common/usb-common.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-core into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-core_5.15.10-1_x86_64.ipk
removed '/home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-audio_5.10.88-1_x86_64.ipk'
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-audio/lib/modules/5.15.10/snd-usb-audio.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-audio/lib/modules/5.15.10/snd-usbmidi-lib.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-audio into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-audio_5.15.10-1_x86_64.ipk
removed '/home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-printer_5.10.88-1_x86_64.ipk'
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-printer/lib/modules/5.15.10/usblp.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-printer into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-printer_5.15.10-1_x86_64.ipk
removed '/home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-storage_5.10.88-1_x86_64.ipk'
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.10/drivers/usb/storage/usb-storage.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-storage_5.15.10-1_x86_64.ipk
removed '/home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-storage-extras_5.10.88-1_x86_64.ipk'
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-usbat.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-sddr09.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-karma.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-cypress.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-sddr55.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-isd200.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-alauda.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-freecom.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-jumpshot.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.15.10/ums-datafab.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-storage-extras_5.15.10-1_x86_64.ipk
removed '/home/builder/lede_x86/bin/targets/x86/64/packages/kmod-usb-storage-uas_5.10.88-1_x86_64.ipk'
[... cut ...]
time: package/kernel/linux/compile#24.96#12.64#36.00
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

GreatMichaelLee, 2021-12-30:
之前一直好好的，x86, 用的5.4内核，今次切换到test kernel(5.10.x)之后就出这个了。

kleisth, 2021-12-30:
5.15内核好像是有问题，你切换到5.15了，还是先用着5.10吧，虽然也有问题。。。。

mitsukileung, 2021-12-30:
5.15内核出现同样的错误

coolsnowwolf, 2021-12-31:
重新配置 .config 。这个在  x86 不再需要

closed 2021-12-31 by coolsnowwolf