coolsnowwolf/lede #8548: kmod-fs-nfs is missing dependencies for the following libraries: nfs_ssc.ko

Nothend, 2021-12-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：kmod-fs-nfs is missing dependencies for the following libraries:nfs_ssc.ko

### (2) 路由器型号和固件版本

A：X86，编译最新固件

### (3) 详细日志

A：Packaged contents of /home/ben/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dnsresolver into /home/ben/lede/bin/targets/x86/64/packages/kmod-dnsresolver_5.10.88-1_x86_64.ipk
removed '/home/ben/lede/bin/targets/x86/64/packages/kmod-fs-nfs_5.10.88-1_x86_64.ipk'
Package kmod-fs-nfs is missing dependencies for the following libraries:
nfs_ssc.ko
make[3]: *** [modules/fs.mk:394: /home/ben/lede/bin/targets/x86/64/packages/kmod-fs-nfs_5.10.88-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/ben/lede/package/kernel/linux'
time: package/kernel/linux/compile#8.45#2.08#10.47
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/ben/lede'
make[1]: *** [package/Makefile:110: /home/ben/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ben/lede'
make: *** [/home/ben/lede/include/toplevel.mk:230: world] Error 2

liwenjie119, 2021-12-31:
同样问题：
make[3]: Leaving directory '/workdir/openwrt/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#1.26#0.76#1.53
make[3]: Entering directory '/workdir/openwrt/package/firmware/prism54-firmware'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/workdir/openwrt/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.21#0.03#0.22
make[3]: Entering directory '/workdir/openwrt/package/kernel/linux'
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-fs-nfs is missing dependencies for the following libraries:
nfs_ssc.ko
make[3]: *** [modules/fs.mk:395: /workdir/openwrt/bin/targets/x86/64/packages/kmod-fs-nfs_5.10.88-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.71#0.19#2.88
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'

Nothend, 2021-12-31:
> 同样问题： make[3]: Leaving directory '/workdir/openwrt/package/firmware/linux-firmware' time: package/firmware/linux-firmware/compile#1.26#0.76#1.53 make[3]: Entering directory [...]

你是不是在menuconfig中选择了luci-app-nfs和luci-app-nft-qos？

coolsnowwolf, 2021-12-31:
fix

Nothend, 2021-12-31:
> 

大雕 ，你好得加个ed啊  哈哈

closed 2021-12-31 by coolsnowwolf