coolsnowwolf/lede #8551: x86编译k5.15内核固件提示缺少依赖：Package kmod-keys-trusted is missing dependencies for the following libraries: asn1_encoder.ko

aakkll, 2021-12-31:
[image]

Package kmod-keys-trusted is missing dependencies for the following libraries:
asn1_encoder.ko

提示是这个依赖

moonerick, 2021-12-31:
问个问题，如何选择5.15的内核进行编译？

aakkll, 2021-12-31:
> 

[image]
k5.15现在是testing_patchver，把内核选择测试版内核就行了

heiger8231, 2021-12-31:
我的也是同样的问题，应该不是个例

kleisth, 2021-12-31:
同样的问题，正想发issue就看到有人发了

Dimitri2020007, 2021-12-31:
我也碰到5.15内核在线编译的问题了

附上日志

 make[1] world
 make[2] target/compile
 make[2] package/cleanup
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 1
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
make[3]: Entering directory '/workdir/openwrt/target/linux'
make[4]: Entering directory '/workdir/openwrt/target/linux/x86'
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/localversion
/workdir/openwrt/scripts/kconfig.pl  + + /workdir/openwrt/target/linux/generic/config-5.15 /workdir/openwrt/target/linux/x86/config-5.15 /workdir/openwrt/target/linux/x86/64/config-5.15 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /workdir/openwrt/.config >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.target
/workdir/openwrt/scripts/package-metadata.pl kconfig /workdir/openwrt/tmp/.packageinfo /workdir/openwrt/.config 5.15 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.override
/workdir/openwrt/scripts/kconfig.pl 'm+' '+' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.target /dev/null /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.override > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set
mv /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.old
grep -v INITRAMFS /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.old > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.prev || { cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config; cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.prev; }
export MAKEFLAGS= ; [ -d /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/user_headers ] || make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Dec 31 02:32:24 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.12 CC="x86_64-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/user_headers headers_install
grep '=[ym]' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.config.set | LC_ALL=C sort | /workdir/openwrt/staging_dir/host/bin/mkhash md5 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.vermagic
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.configured
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/vmlinux /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/System.map
make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Dec 31 02:32:24 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.12 CC="x86_64-openwrt-linux-musl-gcc" bzImage modules
[... cut ...]
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#5.76#1.38#7.08
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:18: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

mmeiwenlong, 2021-12-31:
> > 
> 
> [image] k5.15现在是testing_patchver，把内核选择测试版内核就行了
[image]
我的没有

ohbaby30, 2021-12-31:
同样的错误，make clean甚至删除重来都不行。

WYC-2020, 2021-12-31:
你们还真是追新哈哈，5.10都还没稳定，都5.15了哈哈，5.15肯定问题多暂时还是不要玩5.15，等大佬们更新然后稳定再玩吧

heiger8231, 2021-12-31:
多参与测试，也不是坏事，都不参与了什么时候才能稳定呢😎

coolsnowwolf, 2021-12-31:
什么组件需要这个 kmod-tpm ?

Dimitri2020007, 2021-12-31:
5.15内核都更新了啥？

有对于网络的提升吗？

moonerick, 2021-12-31:
> > Thank you. 默认的是5.10版本了么，5.4版本哪里还能选择么，我的5.10.88没有测试，但是在上一个版本中会出现运行1天左右后dns无法解析的问题，日志也没有报错，必须要重启才能恢复
> 
> [image] k5.15现在是testing_patchver，把内核选择测试版内核就行了

closed 2021-12-31 by coolsnowwolf

kleisth, 2021-12-31:
> 什么组件需要这个 kmod-tpm ?
5.10.x也包含了现在报错的这些组件，但是编译正常，5.15就报错了