coolsnowwolf/lede #8552: 感谢L大这么快修复了asn1-encoder.ko的问题，但是还是编译出错，这次又换了个文件报错

kleisth, 2021-12-31:
SHELL= flock /home/kleist/lede/tmp/.root-copy.flock -c 'cp -fpR /home/kleist/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/kleist/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/kleist/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-mmc-spi is missing dependencies for the following libraries:
of_mmc_spi.ko
make[3]: *** [modules/spi.mk:27: /home/kleist/lede/bin/targets/x86/64/packages/kmod-mmc-spi_5.15.12-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/kleist/lede/package/kernel/linux'
time: package/kernel/linux/compile#1.33#0.07#1.41
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/kleist/lede'
make[1]: *** [package/Makefile:110: /home/kleist/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kleist/lede'
make: *** [/home/kleist/lede/include/toplevel.mk:230：world] 错误 2

----------------------------------------------------------------------------------------
这个文件在5.10.x也是正常编译出来的，在5.15就报错了

Wilson-coder, 2021-12-31:
后面还有个kmod-net-asix报错，把这几个Kmod都取消选择之后，后面还有个poll.h的调用出错......反正后面出的错误已经远远超过了我的能力范围，所以我刚才已经删除了目录，重新回到28号的commit版本了

kleisth, 2021-12-31:
> 后面还有个kmod-net-asix报错，把这几个Kmod都取消选择之后，后面还有个poll.h的调用出错......反正后面出的错误已经远远超过了我的能力范围，所以我刚才已经删除了目录，重新回到28号的commit版本了

如果编译5.10.x内核的话，是没有问题的，你没必要回退，新的5.15.x内核才无法成功编译。

coolsnowwolf, 2021-12-31:
of_mmc_spi 在 x86 不再需要

coolsnowwolf, 2021-12-31:
kmod-mmc-spi  取消选择即可，x86 不存在这个的~

kleisth, 2021-12-31:
> kmod-mmc-spi 取消选择即可，x86 不存在这个的~

好的好的，这个默认钩的，也不知道啥用。而且5.10.x也没报这些问题。。。

Wilson-coder, 2021-12-31:
> > 后面还有个kmod-net-asix报错，把这几个Kmod都取消选择之后，后面还有个poll.h的调用出错......反正后面出的错误已经远远超过了我的能力范围，所以我刚才已经删除了目录，重新回到28号的commit版本了
> 
> 如果编译5.10.x内核的话，是没有问题的，你没必要回退，新的5.15.x内核才无法成功编译。

是的，我明白是新内核可能还有一些代码不太兼容，反正遇到这种Kmod编译出错的，还能大致根据作用判断是否必要，不需要的直接在menuconfig里面取消选择，但是进行到后面开始出现include的代码出错，就完全无能为力了。

closed 2022-01-01 by kleisth