coolsnowwolf/lede #8553: 编译报错，5.10没问题，编译5.15报错

ohbaby30, 2021-12-31:
编译报错，5.10没问题，编译5.15报错

make[6]: Leaving directory '/home/ohbaby/x8615/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/contrib'
make[6]: Entering directory '/home/ohbaby/x8615/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[6]: Leaving directory '/home/ohbaby/x8615/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[5]: *** [Makefile:270: all-recursive] Error 1
make[5]: Leaving directory '/home/ohbaby/x8615/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[4]: *** [Makefile:202: all] Error 2
make[4]: Leaving directory '/home/ohbaby/x8615/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[3]: *** [Makefile:60: /home/ohbaby/x8615/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/.built] Error 2
make[3]: Leaving directory '/home/ohbaby/x8615/lede/package/lean/pdnsd-alt'
time: package/lean/pdnsd-alt/compile#0.24#0.05#0.31
    ERROR: package/lean/pdnsd-alt failed to build.
make[2]: *** [package/Makefile:116: package/lean/pdnsd-alt/compile] Error 1
make[2]: Leaving directory '/home/ohbaby/x8615/lede'
make[1]: *** [package/Makefile:110: /home/ohbaby/x8615/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ohbaby/x8615/lede'
make: *** [/home/ohbaby/x8615/lede/include/toplevel.mk:230：world] 错误 2

Ra2-IFV, 2021-12-31:
用多线程编的？这报错信息也看不出啥
建议make clean 一下重编译
（提一下，咋用5.15啊）

ohbaby30, 2021-12-31:
> 

单线程多线程都报错，而且删除整个文件夹重新gitclone也是一样啊，

两种方法5.15，第一是修改/target/linux/x86下makeflie文件，第二是make menuconfig修改global build setteings-use the testing kernel version打星

martin-xj, 2021-12-31:
一样的问题，整个重新编译后同样出错。

mitsukileung, 2021-12-31:
2021-12-31T10:08:29.1251440Z touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.bnx2x-firmware_installed
2021-12-31T10:08:29.1282221Z echo "bnx2x-firmware" >> /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
2021-12-31T10:08:29.1305095Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/firmware/linux-firmware'
2021-12-31T10:08:29.1315082Z time: package/firmware/linux-firmware/compile#15.17#4.28#18.02
2021-12-31T10:08:29.1458544Z make[3]: Entering directory '/home/runner/work/lede/lede/package/firmware/prism54-firmware'
2021-12-31T10:08:29.3979779Z make[3]: Nothing to be done for 'compile'.
2021-12-31T10:08:29.3981469Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/firmware/prism54-firmware'
2021-12-31T10:08:29.3990989Z time: package/firmware/prism54-firmware/compile#0.21#0.04#0.26
2021-12-31T10:08:29.4130790Z make[3]: Entering directory '/home/runner/work/lede/lede/package/kernel/linux'
2021-12-31T10:08:32.4385451Z mkdir -p /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp
2021-12-31T10:08:32.4386637Z 
2021-12-31T10:08:32.4388615Z SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/'
2021-12-31T10:08:32.4390818Z touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
2021-12-31T10:08:32.4686621Z ERROR: module '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/fs/nfs_common/nfs_ssc.ko' is missing.
2021-12-31T10:08:32.4690821Z make[3]: *** [modules/fs.mk:424: /home/runner/work/lede/lede/bin/targets/x86/64/packages/kmod-fs-nfs-common_5.15.12-1_x86_64.ipk] Error 1
2021-12-31T10:08:32.4692137Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/linux'
2021-12-31T10:08:32.4702681Z time: package/kernel/linux/compile#2.88#0.19#3.06
2021-12-31T10:08:32.4712976Z     ERROR: package/kernel/linux failed to build.
2021-12-31T10:08:32.4721525Z make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
2021-12-31T10:08:32.4748278Z make[2]: Leaving directory '/home/runner/work/lede/lede'
2021-12-31T10:08:32.4757549Z make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-12-31T10:08:32.4775287Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2021-12-31T10:08:32.4801983Z make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 2
2021-12-31T10:08:32.4844389Z ##[error]Process completed with exit code 2.

WYC-2020, 2021-12-31:
[image]
 重复定义而已，应该是5.15的内核已经定义了这个，这个的加一个补丁

WYC-2020, 2021-12-31:
今晚回去修一下应该就可以

Wilson-coder, 2021-12-31:
> [image] 重复定义而已，应该是5.15的内核已经定义了这个，这个的加一个补丁

我也是卡在了这里，看不懂代码，只能等高手修复

baodp, 2021-12-31:
我也一直出这个错，我默认配置只加了个IPV6支持。
群里问大屌，说我菜。他们都编译好了。。。。

WYC-2020, 2021-12-31:
这都是小bug分分钟搞定，netdata最新版我一样搞定了，l大回退了而已，很多东西可以自己修复的，不一定非要等l大，哈哈

souwei168, 2021-12-31:
里面有三个内核心，编译的时候怎么选择 5.10.XX呢
默认好像是 5.4.XXX的

lxl6125, 2021-12-31:
> 里面有三个内核心，编译的时候怎么选择 5.10.XX呢 默认好像是 5.4.XXX的

x86默认就是5.10，其他设备make menuconfig  ---> Global build settings  ---> Use the testing kernel version

mjyhj, 2021-12-31:
5.15正常，大雕说你们菜而已

dfbucy090, 2021-12-31:
> 这都是小bug分分钟搞定，netdata最新版我一样搞定了，l大回退了而已，很多东西可以自己修复的，不一定非要等l大，哈哈

我信您，他们回退后使用默认编译，32位的netdata还是不行，用您的方法可以通过。等待您修复5.15的消息。

WYC-2020, 2021-12-31:
你不编译pdnsd一样不会遇到这个问题，只是每个人编译的模块不一样而已，能过不一定别人就能过哈哈

zhengyiquan, 2021-12-31:
> > 里面有三个内核心，编译的时候怎么选择 5.10.XX呢 默认好像是 5.4.XXX的
> 
> x86默认就是5.10，其他设备make menuconfig ---> Global build settings ---> Use the testing kernel version

可以回退5.4吗？我想编译koolproxy进去koolproxy 不支持5.1内核，开启插件频繁报错

lxl6125, 2022-01-01:
> > > 里面有三个内核心，编译的时候怎么选择 5.10.XX呢 默认好像是 5.4.XXX的
> > 
> > 
> > x86默认就是5.10，其他设备make menuconfig ---> Global build settings ---> Use the testing kernel version
> 
> 可以回退5.4吗？我想编译koolproxy进去koolproxy 不支持5.1内核，开启插件频繁报错

改 target/linux/x86/Makefile

zhengyiquan, 2022-01-01:
> > > > 里面有三个内核心，编译的时候怎么选择 5.10.XX呢 默认好像是 5.4.XXX的
> 
> > > 
> 
> > > 
> 
> > > x86默认就是5.10，其他设备make menuconfig ---> Global build settings ---> Use the testing kernel version
> 
> > 
> 
> > 可以回退5.4吗？我想编译koolproxy进去koolproxy 不支持5.1内核，开启插件频繁报错
> 
> 
> 
> 改 target/linux/x86/Makefile
>好的，我试试，感谢

closed 2022-01-02 by coolsnowwolf