coolsnowwolf/lede #8556: X86默认配置编译出错

baodp, 2021-12-31:
time: package/lean/r8125/compile#3.55#0.25#3.89
  CC [M]  /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/src/r8168_asf.o
  CC [M]  /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/src/rtl_eeprom.o
  CC [M]  /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/src/rtltool.o
  LD [M]  /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/src/r8168.o
  MODPOST /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/src/Module.symvers
  LD [M]  /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/src/r8168.ko
make[5]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.88'
make[4]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/src'
for subdir in .; do realdir=$(readlink -f /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02); grep -F /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02 /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/$subdir/Module.symvers >> /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/Module.symvers.tmp; [ "/home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02" = "$realdir" ] || grep -F $realdir /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/$subdir/Module.symvers >> /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/Module.symvers.tmp; done; sort -u /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/Module.symvers.tmp > /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/Module.symvers; mkdir -p /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers; mv /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/Module.symvers /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers
grep: /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/./Module.symvers: No such file or directory
touch /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/.built
rm -rf /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/.pkgdir/kmod-r8168.installed /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/.pkgdir/kmod-r8168
mkdir -p /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/.pkgdir/kmod-r8168
mkdir -p /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/.pkgdir/kmod-r8168/etc/modules.d; ( echo "r8168"; ) > /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/.pkgdir/kmod-r8168/etc/modules.d/r8168;
touch /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/.pkgdir/kmod-r8168.installed
mkdir -p /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/bdp/lede/tmp/.root-copy.flock -c 'cp -fpR /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/.pkgdir/kmod-r8168/. /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-r8168_installed
removed '/home/bdp/lede/bin/targets/x86/64/packages/kmod-r8168_5.10.88+8.049.02-x86-_x86_64.ipk'
mkdir -p /home/bdp/lede/bin/targets/x86/64/packages /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168/CONTROL /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo
mkdir -p /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168/etc/modules.d; ( echo "r8168"; ) > /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168/etc/modules.d/r8168;
find /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168 -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/bdp/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/bdp/lede/scripts/strip-kmod.sh" PATCHELF="/home/bdp/lede/staging_dir/host/bin/patchelf" /home/bdp/lede/scripts/rstrip.sh /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168
rstrip.sh: /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168/lib/modules/5.10.88/r8168.ko: relocatable
(cd /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/bdp/lede/bin/targets/x86/64/packages
/home/bdp/lede/staging_dir/host/bin/fakeroot /home/bdp/lede/scripts/ipkg-build -m "" /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168 /home/bdp/lede/bin/targets/x86/64/packages
Packaged contents of /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.02/ipkg-x86_64/kmod-r8168 into /home/bdp/lede/bin/targets/x86/64/packages/kmod-r8168_5.10.88+8.049.02-x86-_x86_64.ipk
echo "kmod-r8168" >> /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo/r8168.default.install
make[3]: Leaving directory '/home/bdp/lede/package/lean/r8168'
time: package/lean/r8168/compile#9.03#0.28#9.41
make[2]: Leaving directory '/home/bdp/lede'
make[1]: *** [package/Makefile:110: /home/bdp/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bdp/lede'
make: *** [/home/bdp/lede/include/toplevel.mk:230: world] Error 2

上面是出错最后日志。选测试版也这样。求教下什么问题

baodp, 2021-12-31:
我试了单线程也编译不出来。我太菜了。

WYC-2020, 2021-12-31:
那就贴单线程的错误日志

ccblinda, 2021-12-31:
我重新拉代码成功了

baodp, 2021-12-31:
> 那个贴单线程的错误日志

make[8]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc/SuSE'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc/SuSE'
Making all in Debian
make[8]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc/Debian'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc/Debian'
Making all in Slackware
make[8]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc/Slackware'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc/Slackware'
Making all in ArchLinux
make[8]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc/ArchLinux'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc/ArchLinux'
make[8]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc'
make[8]: Nothing to be done for 'all-am'.
make[8]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc'
make[7]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc'
Making all in test
make[7]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/test'
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../..   -I/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par=pdnsd-1.2.9b-par -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/bdp/lede/staging_dir/target-x86_64_musl/usr/include  -pthread -c ../icmp.c
In file included from ../icmp.c:31:
/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^~~~~~~
In file included from ../dns.h:30,
                 from ../cache.h:31,
                 from ../helpers.h:32,
                 from ../error.h:35,
                 from ../icmp.c:63:
/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/net/if.h:64:8: error: redefinition of 'struct ifmap'
 struct ifmap {
        ^~~~~
In file included from /home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/icmp.h:23,
                 from ../icmp.c:44:
/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/if.h:194:8: note: originally defined here
 struct ifmap {
        ^~~~~
In file included from ../dns.h:30,
                 from ../cache.h:31,
                 from ../helpers.h:32,
                 from ../error.h:35,
                 from ../icmp.c:63:
/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/net/if.h:76:8: error: redefinition of 'struct ifreq'
 struct ifreq {
        ^~~~~
In file included from /home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/icmp.h:23,
                 from ../icmp.c:44:
/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/if.h:232:8: note: originally defined here
 struct ifreq {
        ^~~~~
In file included from ../dns.h:30,
                 from ../cache.h:31,
                 from ../helpers.h:32,
                 from ../error.h:35,
                 from ../icmp.c:63:
/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/net/if.h:116:8: error: redefinition of 'struct ifconf'
 struct ifconf {
        ^~~~~~
In file included from /home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/icmp.h:23,
                 from ../icmp.c:44:
/home/bdp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/if.h:284:8: note: originally defined here
 struct ifconf  {
        ^~~~~~
make[7]: *** [Makefile:460: icmp.o] Error 1
make[7]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/test'
make[6]: *** [Makefile:607: all-recursive] Error 1
make[6]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src'
Making all in doc
make[6]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/doc'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/doc'
Making all in contrib
make[6]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/contrib'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/contrib'
make[6]: Entering directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[6]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[5]: *** [Makefile:270: all-recursive] Error 1
make[5]: Leaving directory '/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
[... cut ...]
make[3]: Leaving directory '/home/bdp/lede/package/lean/pdnsd-alt'
time: package/lean/pdnsd-alt/compile#0.16#0.00#0.15
    ERROR: package/lean/pdnsd-alt failed to build.
make[2]: *** [package/Makefile:116: package/lean/pdnsd-alt/compile] Error 1
make[2]: Leaving directory '/home/bdp/lede'
make[1]: *** [package/Makefile:110: /home/bdp/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bdp/lede'
make: *** [/home/bdp/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2021-12-31:
另外一个帖子又说为什么出错，后面在修复一下

tianiue, 2021-12-31:
/home/bdp/lede/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par
我也是这里报错

baodp, 2021-12-31:
arker lsefimmap.marker lsefi.marker efifwsetup.marker blocklist.marker boot.marker cat.marker cmp.marker configfile.marker cpuid.marker date.marker echo.marker eval.marker extcmd.marker fixvideo.marker gptsync.marker halt.marker reboot.marker hashsum.marker pgp.marker hdparm.marker help.marker hexdump.marker keystatus.marker loadbios.marker loadenv.marker ls.marker lsmmap.marker lspci.marker memrw.marker minicmd.marker parttool.marker password.marker password_pbkdf2.marker play.marker spkmodem.marker morse.marker probe.marker read.marker search.marker search_fs_file.marker search_fs_uuid.marker search_label.marker setpci.marker pcidump.marker sleep.marker smbios.marker terminal.marker test.marker true.marker usbtest.marker videoinfo.marker videotest.marker xnu_uuid.marker dm_nv.marker loopback.marker cryptodisk.marker json.marker afsplitter.marker luks.marker luks2.marker geli.marker lvm.marker ldm.marker mdraid09.marker mdraid09_be.marker mdraid1x.marker diskfilter.marker raid5rec.marker raid6rec.marker scsi.marker memdisk.marker ata.marker ahci.marker pata.marker usbms.marker font.marker procfs.marker affs.marker afs.marker bfs.marker zstd.marker btrfs.marker archelp.marker cbfs.marker cpio.marker cpio_be.marker newc.marker odc.marker ext2.marker fat.marker exfat.marker f2fs.marker fshelp.marker hfs.marker hfsplus.marker hfspluscomp.marker iso9660.marker jfs.marker minix.marker minix2.marker minix3.marker minix_be.marker minix2_be.marker minix3_be.marker nilfs2.marker ntfs.marker ntfscomp.marker reiserfs.marker romfs.marker sfs.marker squash4.marker tar.marker udf.marker ufs1.marker ufs1_be.marker ufs2.marker xfs.marker zfs.marker zfscrypt.marker zfsinfo.marker macbless.marker gettext.marker gfxmenu.marker hello.marker gzio.marker offsetio.marker bufio.marker elf.marker crypto.marker pbkdf2.marker relocator.marker datetime.marker setjmp.marker aout.marker bsd.marker linux16.marker multiboot2.marker multiboot.marker linux.marker xnu.marker random.marker macho.marker appleldr.marker chain.marker mmap.marker normal.marker part_acorn.marker part_amiga.marker part_apple.marker part_gpt.marker part_msdos.marker part_sun.marker part_plan.marker part_dvh.marker part_bsd.marker part_sunpc.marker part_dfly.marker msdospart.marker at_keyboard.marker gfxterm.marker gfxterm_background.marker serial.marker terminfo.marker usb_keyboard.marker video_cirrus.marker video_bochs.marker functional_test.marker exfctest.marker strtoull_test.marker setjmp_test.marker signature_test.marker sleep_test.marker xnu_uuid_test.marker pbkdf2_test.marker legacy_password_test.marker div.marker div_test.marker mul_test.marker shift_test.marker cmp_test.marker ctz_test.marker bswap_test.marker videotest_checksum.marker gfxterm_menu.marker cmdline_cat_test.marker bitmap.marker bitmap_scale.marker efi_gop.marker efi_uga.marker jpeg.marker png.marker tga.marker video_fb.marker video.marker video_colors.marker datehook.marker net.marker tftp.marker http.marker efinet.marker legacycfg.marker syslinuxcfg.marker test_blockarg.marker xzio.marker lzopio.marker testload.marker backtrace.marker keylayouts.marker priority_queue.marker time.marker adler32.marker crc64.marker mpi.marker all_video.marker testspeed.marker tpm.marker tr.marker progress.marker file.marker rdmsr.marker wrmsr.marker gcry_arcfour.marker gcry_blowfish.marker gcry_camellia.marker gcry_cast5.marker gcry_crc.marker gcry_des.marker gcry_dsa.marker gcry_idea.marker gcry_md4.marker gcry_md5.marker gcry_rfc2268.marker gcry_rijndael.marker gcry_rmd160.marker gcry_rsa.marker gcry_seed.marker gcry_serpent.marker gcry_sha1.marker gcry_sha256.marker gcry_sha512.marker gcry_tiger.marker gcry_twofish.marker gcry_whirlpool.marker; do \
  b=`basename $pp .marker`; \
  sed -n \
    -e "/EXTCOMMAND_LIST_MARKER *( *\"/{s/.*( *\"\([^\"]*\)\".*/*\1: $b/;p;}" \
    -e "/EXTCOMMAND_LOCKDOWN_LIST_MARKER *( *\"/{s/.*( *\"\([^\"]*\)\".*/*\1: $b/;p;}" \
    -e "/P1COMMAND_LIST_MARKER *( *\"/{s/.*( *\"\([^\"]*\)\".*/*\1: $b/;p;}" \
    -e "/COMMAND_LIST_MARKER *( *\"/{s/.*( *\"\([^\"]*\)\".*/\1: $b/;p;}" \
    -e "/COMMAND_LOCKDOWN_LIST_MARKER *( *\"/{s/.*( *\"\([^\"]*\)\".*/\1: $b/;p;}" $pp; \
done) | sort -u > command.lst
(for pp in disk.marker trig.marker cs5536.marker usb.marker usbserial_common.marker usbserial_pl2303.marker usbserial_ftdi.marker usbserial_usbdebug.marker uhci.marker ohci.marker ehci.marker nativedisk.marker iorw.marker cbtable.marker cbtime.marker cbls.marker cbmemc.marker regexp.marker acpi.marker lsacpi.marker lsefisystab.marker lssal.marker lsefimmap.marker lsefi.marker efifwsetup.marker blocklist.marker boot.marker cat.marker cmp.marker configfile.marker cpuid.marker date.marker echo.marker eval.marker extcmd.marker fixvideo.marker gptsync.marker halt.marker reboot.marker hashsum.marker pgp.marker hdparm.marker help.marker hexdump.marker keystatus.marker loadbios.marker loadenv.marker ls.marker lsmmap.marker lspci.marker memrw.marker minicmd.marker parttool.marker password.marker password_pbkdf2.marker play.marker spkmodem.marker morse.marker probe.marker read.marker search.marker search_fs_file.marker search_fs_uuid.marker search_label.marker setpci.marker pcidump.marker sleep.marker smbios.marker terminal.marker test.marker true.marker usbtest.marker videoinfo.marker videotest.marker xnu_uuid.marker dm_nv.marker loopback.marker cryptodisk.marker json.marker afsplitter.marker luks.marker luks2.marker geli.marker lvm.marker ldm.marker mdraid09.marker mdraid09_be.marker  [... cut ...]
make[3]: Leaving directory '/home/bdp/lede/package/lean/pdnsd-alt'
time: package/lean/pdnsd-alt/compile#0.13#0.03#0.15
    ERROR: package/lean/pdnsd-alt failed to build.
make[2]: *** [package/Makefile:116: package/lean/pdnsd-alt/compile] Error 1
make[2]: Leaving directory '/home/bdp/lede'
make[1]: *** [package/Makefile:110: /home/bdp/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bdp/lede'
make: *** [/home/bdp/lede/include/toplevel.mk:230: world] Error 2

baodp, 2021-12-31:
等待阿祖更新了

closed 2022-01-01 by coolsnowwolf