coolsnowwolf/lede #8558: 5.15 X86 编译出错 nfs_ssc.ko is missing

mitsukileung, 2021-12-31:
X86

SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
ERROR: module '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/fs/nfs_common/nfs_ssc.ko' is missing.
make[3]: *** [modules/fs.mk:424: /home/runner/work/lede/lede/bin/targets/x86/64/packages/kmod-fs-nfs-common_5.15.12-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.62#0.19#2.80
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

zhengyiquan, 2021-12-31:
哎，一转5.1就一堆人报错

harleywong, 2022-01-01:
同样错误

coolsnowwolf, 2022-01-01:
你们需要重新生成 .config

closed 2022-01-01 by coolsnowwolf

GreatMichaelLee, 2022-01-01:
> 你们需要重新生成 .config

不想重新生成.config,太麻烦了，之前定制化选了一堆插件,lib和网络工具，重新生成得全没了，还得重新勾，能否说下去掉哪个能fix这个error? @coolsnowwolf