coolsnowwolf/lede #8560: E8350 v1固件编译成功，无法刷入

cevinwu87, 2022-01-01:
反馈bug/问题模板，提建议请删除

## 1.E8350v1固件能成功编译，但是无论在op下刷ubi文件还是不死下刷bin文件都无法刷入

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：新编译的固件不死刷入提供image验证不通过

ERROR - the image you uploaded failed to pass verification.

B：在op下刷入还是原来的固件

### (2) 路由器型号和固件版本

A：E8350 v1，原厂系统已经刷到1.0.03，群里面7月份的op和Pandorabox能正常刷入

### (3) 详细日志

A：

coolsnowwolf, 2022-01-01:
先用7月份的固件，然后 web 升级到最新版就行了

cevinwu87, 2022-01-02:
已经尝试过多次，web刷入后还是原来的固件，新的固件并没有刷入

分区信息
dev:    size   erasesize  name
mtd0: 04000000 00020000 "ubi"
mtd1: 04000000 00020000 "extra"
mtd2: 00400000 00010000 "spi0.0"

内核日志
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.132 (runner@fv-az163-767) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0-737e0fd)) #0 SMP Fri Jul 23 13:09:30 2021
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Linksys E8350 V1 WiFi Router
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 122880
[    0.000000]   Normal zone: 1080 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 476880K/491520K available (6256K kernel code, 198K rwdata, 1568K rodata, 1024K init, 232K bss, 14640K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000006] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000245] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000265] pid_max: default: 32768 minimum: 301
[    0.000475] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000492] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001569] CPU: Testing write buffer coherency: ok
[    0.002730] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002911] rcu: Hierarchical SRCU implementation.
[    0.003421] smp: Bringing up secondary CPUs ...
[    0.005382] smp: Brought up 1 node, 2 CPUs
[    0.005399] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005410] CPU: All CPU(s) started in SVC mode.
[    0.016335] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.016505] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016536] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.016682] pinctrl core: initialized pinctrl subsystem
[    0.017840] NET: Registered protocol family 16
[    0.018129] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.019546] cpuidle: using governor ladder
[    0.019613] cpuidle: using governor menu
[    0.056955] usbcore: registered new interface driver usbfs
[    0.057036] usbcore: registered new interface driver hub
[    0.057106] usbcore: registered new device driver usb
[    0.057165] pps_core: LinuxPPS API ver. 1 registered
[    0.057178] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.057219] PTP clock support registered
[    0.058109] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.059278] clocksource: Switched to clocksource dg_timer
[    0.060180] thermal_sys: Registered thermal governor 'step_wise'
[    0.062857] NET: Registered protocol family 2
[    0.063016] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.063883] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.063923] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[... 442 lines cut ...]
[  112.027554] ath10k_pci 0001:01:00.0: wmi-event-beacon-tx, could not find vdev for id: 0
[  115.052974] ath10k_pci 0001:01:00.0: WARNING: failed to wait for beacon tx callback for vdev 0: 0
[  115.282180] wlan0: Destroyed NSS virtual interface
[  115.282214] wlan0: RX-Q purged.
[  115.862576] debugfs: File 'virt_if' in directory 'stats' already present!
[  115.862921] wlan0: Created a NSS virtual interface.
[  115.868343] wlan0: RX-Q and tasklet initialized.
[  116.880801] ath10k_pci 0001:01:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[  116.890174] ath10k_pci 0001:01:00.0: wmi print 'P 128 V 8 T 410'
[  116.890443] ath10k_pci 0001:01:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[  116.895862] ath10k_pci 0001:01:00.0: wmi print 'alloc rem: 24984 iram: 38672'
[  116.966599] ath10k_pci 0001:01:00.0: pdev param 0 not supported by firmware
[... cut ...]
[  117.005192] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  117.005466] br-lan: port 2(wlan0) entered blocking state
[  117.010748] br-lan: port 2(wlan0) entered forwarding state
[  123.265961] br-lan: port 3(wifi0) entered blocking state
[  123.266020] br-lan: port 3(wifi0) entered disabled state
[  123.270907] device wifi0 entered promiscuous mode
[  123.275634] device host0 entered promiscuous mode
[  123.280593] br-lan: port 3(wifi0) entered blocking state
[  123.284930] br-lan: port 3(wifi0) entered forwarding state

davidtall, 2023-05-18:
same issue

coolsnowwolf, 2026-09-05:
已修复并推送提交 `9a5c57a1b1c`：恢复 Linksys E8350 v1 支持，加入正确的 DTS/NAND 分区、网络与 MAC 初始化、升级路径，以及 Linksys Web 所需的 factory 镜像头和设备 ID 生成流程。该修复针对原 issue 的“编译成功但 Web 验证失败/刷入后仍是旧固件”问题，关闭 issue。

closed 2026-09-05 by coolsnowwolf