coolsnowwolf/lede #8563: 使用Github Actions编译的固件如何用内核版本号命名

lon91ong, 2022-01-01:
编译脚本参考[P3TERX大侠的项目](https://github.com/P3TERX/Actions-OpenWrt)
目前编译的固件只有刷入后才能知道内核版本号
有没有办法编译完后在文件名中加入内核版本号呢?

Jason6111, 2022-01-01:
[image]
tag改下名称不就得了

kleisth, 2022-01-02:
云编译的我不知道，我本地编译的直接改image.mk的IMG_PREFIX那里，加个$(LINUX_VERSION)就行了。。。随便还加了个时间显示。

Jason6111, 2022-01-02:
在脚本中发布固件用mv重命名下也可以 哈哈哈

Jason6111, 2022-01-03:
[image]
[image]
mv a b 就可以

binge8, 2022-01-04:
看我的actions编译的重命名效果
https://github.com/binge8/op/releases
[image]
只需修改这一个地方。

lon91ong, 2022-01-06:
参考 @kleisth 大佬的建议, 最后使用下面的方案解决:
`sed -i ':a;N;$!ba;s/$(BOARD)/&-$(LINUX_VERSION)/1' include/image.mk`

closed 2022-01-06 by lon91ong