coolsnowwolf/lede #8564: 5.15 加了插件就错误了

jaychin2, 2022-01-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5.15 加了插件就错误了

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：-- The C compiler identification is GNU 9.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/x86openwrt/新版内核/lede/staging_dir/host/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ZLIB: /home/x86openwrt/新版内核/lede/staging_dir/host/lib/libz.a (found version "1.2.11") 
CMake Error at /home/x86openwrt/新版内核/lede/staging_dir/host/share/cmake-3.22/Modules/FindPkgConfig.cmake:123 (set):
  Syntax error in cmake code at

    /home/x86openwrt/新版内核/lede/staging_dir/host/share/cmake-3.22/Modules/FindPkgConfig.cmake:123

  when parsing string

    ;/home/x86openwrt/\�\�\�\�\�\�\�\�\�\�\�\�/lede/staging_dir/host/lib

  Invalid escape sequence \�
Call Stack (most recent call first):
  /home/x86openwrt/新版内核/lede/staging_dir/host/share/cmake-3.22/Modules/FindPkgConfig.cmake:150 (_pkgconfig_set)
  /home/x86openwrt/新版内核/lede/staging_dir/host/share/cmake-3.22/Modules/FindPkgConfig.cmake:165 (_pkgconfig_invoke)
  /home/x86openwrt/新版内核/lede/staging_dir/host/share/cmake-3.22/Modules/FindPkgConfig.cmake:637 (_pkgconfig_invoke_dyn)
  /home/x86openwrt/新版内核/lede/staging_dir/host/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  /home/x86openwrt/新版内核/lede/staging_dir/host/share/cmake-3.22/Modules/FindOpenSSL.cmake:134 (pkg_check_modules)
  CMakeLists.txt:6 (INCLUDE)

-- Configuring incomplete, errors occurred!
See also "/home/x86openwrt/新版内核/lede/build_dir/host/firmware-utils-2021-10-05-f9ad6b33/CMakeFiles/CMakeOutput.log".
make[3]: *** [Makefile:22: /home/x86openwrt/新版内核/lede/build_dir/host/firmware-utils-2021-10-05-f9ad6b33/.configured] Error 1
make[3]: Leaving directory '/home/x86openwrt/新版内核/lede/tools/firmware-utils'
time: tools/firmware-utils/compile#0.22#0.10#0.30
    ERROR: tools/firmware-utils failed to build.
make[2]: *** [tools/Makefile:160: tools/firmware-utils/compile] Error 1
make[2]: Leaving directory '/home/x86openwrt/新版内核/lede'
make[1]: *** [tools/Makefile:156: /home/x86openwrt/新版内核/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/x86openwrt/新版内核/lede'
make: *** [/home/x86openwrt/新版内核/lede/include/toplevel.mk:230：world] 错误 2

jaychin2, 2022-01-01:
我是本地编译的，直接编译5.15可以成功，加了kenzok8的passwall和openclash插件之后就失败

lxl6125, 2022-01-01:
openclash我直接用原作者仓库的，没问题，passwall我没试，但是我加无线网卡驱动报错

WYC-2020, 2022-01-01:
没事搞什么中文路径

jaychin2, 2022-01-01:
那我改成英文试试看

jaychin2, 2022-01-01:
好像是第三方源的问题

closed 2022-01-02 by coolsnowwolf