coolsnowwolf/lede #8569: 可道云在5.15内核下无法工作

hxlls, 2022-01-01:
### (1) 具体问题

A：可道云在5.15的内核下，php 提示未运行，是PHP不兼容新的内核还是可道云的问题？

### (2) 路由器型号和固件版本

A：X86-64

### (3) 详细日志

A：内核和系统日志没对应的日志产生

hxlls, 2022-01-01:
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'curl.so' (tried: /usr/lib/php/curl.so (Error loading shared library /usr/lib/php/curl.so: No such file or directory), /usr/lib/php/curl.so.so (Error loading shared library /usr/lib/php/curl.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'dom.so' (tried: /usr/lib/php/dom.so (Error loading shared library /usr/lib/php/dom.so: No such file or directory), /usr/lib/php/dom.so.so (Error loading shared library /usr/lib/php/dom.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'gd.so' (tried: /usr/lib/php/gd.so (Error loading shared library /usr/lib/php/gd.so: No such file or directory), /usr/lib/php/gd.so.so (Error loading shared library /usr/lib/php/gd.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'iconv.so' (tried: /usr/lib/php/iconv.so (Error loading shared library /usr/lib/php/iconv.so: No such file or directory), /usr/lib/php/iconv.so.so (Error loading shared library /usr/lib/php/iconv.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'mbstring.so' (tried: /usr/lib/php/mbstring.so (Error loading shared library /usr/lib/php/mbstring.so: No such file or directory), /usr/lib/php/mbstring.so.so (Error loading shared library /usr/lib/php/mbstring.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'mysqlnd.so' (tried: /usr/lib/php/mysqlnd.so (Error loading shared library /usr/lib/php/mysqlnd.so: No such file or directory), /usr/lib/php/mysqlnd.so.so (Error loading shared library /usr/lib/php/mysqlnd.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'pdo.so' (tried: /usr/lib/php/pdo.so (Error loading shared library /usr/lib/php/pdo.so: No such file or directory), /usr/lib/php/pdo.so.so (Error loading shared library /usr/lib/php/pdo.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'pdo_mysql.so' (tried: /usr/lib/php/pdo_mysql.so (Error loading shared library /usr/lib/php/pdo_mysql.so: No such file or directory), /usr/lib/php/pdo_mysql.so.so (Error loading shared library /usr/lib/php/pdo_mysql.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'pdo_sqlite.so' (tried: /usr/lib/php/pdo_sqlite.so (Error loading shared library /usr/lib/php/pdo_sqlite.so: No such file or directory), /usr/lib/php/pdo_sqlite.so.so (Error loading shared library /usr/lib/php/pdo_sqlite.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'session.so' (tried: /usr/lib/php/session.so (Error loading shared library /usr/lib/php/session.so: No such file or directory), /usr/lib/php/session.so.so (Error loading shared library /usr/lib/php/session.so.so: No such file or directory)) in Unknown on line 0
Sat Jan  1 23:11:13 2022 daemon.err uhttpd[3670]: [01-Jan-2022 23:11:13] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'sqlite3.so' (tried: /usr/lib/php/sqlite3.so (Error loading shared library /usr/lib/php/sqlite3.so: No such file or directory), /usr/lib/php/sqlite3.so.so (Error loading shared library /usr/lib/php/sqlite3.so.so: No such file or directory)) in Unknown on line 0

以上为手动启动PHP后系统产生的日志，但是我到对应目录下能找到对应的文件

WYC-2020, 2022-01-02:
你没看日志加载的路径啊，明显就不是php8

hxlls, 2022-01-02:
[image]
之前眼花，在上去看了次，没php 这个目录。。。

WYC-2020, 2022-01-07:
[image]
这里改下试下，怕不是但在5.15上不行吧 @hxlls

WYC-2020, 2022-01-08:
https://github.com/WYC-2020/lede/commit/0eac96ac86f6fbbfa54d13088fb3b05e7809fe16
看这个

coolsnowwolf, 2026-09-06:
可道云 PHP 兼容性没有 PHP 版本、扩展列表或应用日志，无法归因 5.15 内核，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf