coolsnowwolf/lede #8571: 5.15编译报错

596142041, 2022-01-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译报错

### (2) 路由器型号和固件版本

A：x86,5.15

### (3) 详细日志

A：Installing helper programs:       /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/bin/
  program: phpize
  program: php-config
Installing man pages:             /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/man/man1/
  page: phpize.1
  page: php-config.1
make[4]: Leaving directory '/home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1'
rm -rf /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/{share,man,sbin}
rm -f /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/bin/{php,php-cgi,php-cli}
mv /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/bin/phpize /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/bin/phpize8
mv /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/bin/php-config /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/bin/php8-config
mv /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/include/php /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/include/php8
mv /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/lib/php /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr/lib/php8
cp -fpR /home/lpr/lede/build_dir/target-x86_64_musl/php-8.1.1/staging/usr /home/lpr/lede/staging_dir/target-x86_64_musl/
sed -i -e "s#prefix='/usr'#prefix='/home/lpr/lede/staging_dir/target-x86_64_musl/usr'#" /home/lpr/lede/staging_dir/target-x86_64_musl/usr/bin/phpize8
sed -i -e "s#exec_prefix=\"\`eval echo /usr\`\"#exec_prefix='/home/lpr/lede/staging_dir/target-x86_64_musl/usr'#" /home/lpr/lede/staging_dir/target-x86_64_musl/usr/bin/phpize8
sed -i -e "s#/include\`/php\"#/include\`/php8\"#" /home/lpr/lede/staging_dir/target-x86_64_musl/usr/bin/phpize8
sed -i -e "s#/lib/php\`/build\"#/lib/php8\`/build\"#" /home/lpr/lede/staging_dir/target-x86_64_musl/usr/bin/phpize8
sed -i -e "s#prefix=\"/usr\"#prefix=\"/home/lpr/lede/staging_dir/target-x86_64_musl/usr\"#" /home/lpr/lede/staging_dir/target-x86_64_musl/usr/bin/php8-config
sed -i -e "s#/include/php\"#/include/php8\"#" /home/lpr/lede/staging_dir/target-x86_64_musl/usr/bin/php8-config
find /home/lpr/lede/tmp/stage-php8 -name '*.la' | xargs -r rm -f; 
if [ -f /home/lpr/lede/staging_dir/target-x86_64_musl/packages/php8.list ]; then /home/lpr/lede/scripts/clean-package.sh "/home/lpr/lede/staging_dir/target-x86_64_musl/packages/php8.list" "/home/lpr/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/lpr/lede/tmp/stage-php8 ]; then (cd /home/lpr/lede/tmp/stage-php8; find ./ > /home/lpr/lede/tmp/stage-php8.files); 	SHELL= flock /home/lpr/lede/tmp/.staging-dir.flock -c ' mv /home/lpr/lede/tmp/stage-php8.files /home/lpr/lede/staging_dir/target-x86_64_musl/packages/php8.list && cp -fpR /home/lpr/lede/tmp/stage-php8/* /home/lpr/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/lpr/lede/tmp/stage-php8
touch /home/lpr/lede/staging_dir/target-x86_64_musl/stamp/.php8_installed
make[3]: Leaving directory '/home/lpr/lede/feeds/packages/lang/php8'
time: package/feeds/packages/php8/compile#122.79#21.96#101.04
make[2]: Leaving directory '/home/lpr/lede'
make[1]: *** [package/Makefile:110: /home/lpr/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lpr/lede'
make: *** [/home/lpr/lede/include/toplevel.mk:230：world] 错误 2
多次编译都是同样的错误
,make clean再次编译还是同样的错误

WYC-2020, 2022-01-02:
我都已经开始用起了，可能还是某个插件和5.15有问题

kleisth, 2022-01-02:
之前我也是各种报错，后面全删了重新拉源码编译后问题都没了

dfbucy090, 2022-01-02:
> 我都已经开始用起了，可能还是某个插件和5.15有问题

5.15是不是不支持32位？我编译了64位可以，32位报错。

WYC-2020, 2022-01-02:
应该不会才对，有时间我编译看下才知道，我没编译过5.15 32位的 @dfbucy090

dfbucy090, 2022-01-02:
[image]

coolsnowwolf, 2022-01-02:
5.15 目前暂时不考虑 32bit 的 x86 了，不过还是可以自行编译，遇到有选项选择回车即可

dfbucy090, 2022-01-02:
> 5.15 目前暂时不考虑 32bit 的 x86 了，不过还是可以自行编译，遇到有选项选择回车即可

谢谢大佬回复，重新编译了一下，选择处回了10多个车，还是出错了
[image]
。

dfbucy090, 2022-01-03:
> 5.15 目前暂时不考虑 32bit 的 x86 了，不过还是可以自行编译，遇到有选项选择回车即可

又编译了一遍，32位可以了，谢谢

closed 2022-01-04 by 596142041