coolsnowwolf/lede #8576: 5.15编译报错，kmod-mmc-spi缺少依赖

coldrook, 2022-01-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：kmod-mmc-spi缺少依赖

### (2) 路由器型号和固件版本

A：X86 5.15

### (3) 详细日志

A：Package kmod-mmc-spi is missing dependencies for the following libraries:
of_mmc_spi.ko
make[3]: *** [modules/spi.mk:27: /workdir/openwrt/bin/targets/x86/64/packages/kmod-mmc-spi_5.15.12-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.57#0.22#2.78
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

Jason6111, 2022-01-02:
删掉原来.config 重新配置试下

kleisth, 2022-01-02:
最好是干脆全删了重新拉源码编译，我就是这样搞了之后就没再报错了

coldrook, 2022-01-03:
.config重新配置后解决问题

closed 2022-01-03 by coldrook