coolsnowwolf/lede #8580: 5.15编译错误，已经困扰2天了。

jaychin2, 2022-01-02:
大佬救命！
编译5.15内核的问题
加了src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git small https://github.com/kenzok8/small
这两个里面的插件之后就会出现（不加编译就会成功）
weighted random twos choice least-connection scheduling (IP_VS_TWOS) [N/m/?] (NEW) 
无论选N、M或者回车都会编译失败，Y不让选

错误日志：
make[5]: *** [scripts/Makefile.build:277: /home/x86openwrt/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.o] Error 1
make[4]: *** [Makefile:1868: /home/x86openwrt/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe] Error 2
make[4]: Leaving directory '/home/x86openwrt/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
make[3]: *** [Makefile:87: /home/x86openwrt/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built] Error 2
make[3]: Leaving directory '/home/x86openwrt/openwrt/lede/package/lean/shortcut-fe/shortcut-fe'
time: package/lean/shortcut-fe/shortcut-fe/compile#0.95#0.34#1.29
    ERROR: package/lean/shortcut-fe/shortcut-fe failed to build.
make[2]: *** [package/Makefile:116: package/lean/shortcut-fe/shortcut-fe/compile] Error 1
make[2]: Leaving directory '/home/x86openwrt/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/x86openwrt/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86openwrt/openwrt/lede'
make: *** [/home/x86openwrt/openwrt/lede/include/toplevel.mk:230：world] 错误 2

coolsnowwolf, 2022-01-02:
5.15 不支持 SFE

jaychin2, 2022-01-02:
感谢L大，我马上去试

jaychin2, 2022-01-02:
大佬你说的SFE是不是Turbo ACC里的SFE，我去掉了之后编译仍然让我做选择题。
weighted random twos choice least-connection scheduling (IP_VS_TWOS) [N/m/?] (NEW)

kleisth, 2022-01-02:
全删了重新拉源码，我和你用的一样第三方插件库，开始也是不停出错编译不了

jaychin2, 2022-01-02:
重拉了起码6遍了，只有不用三方才能成功。

kleisth, 2022-01-02:
是从头重新配置的吗？拉下来后用默认编译一次5.15的，看行不行，编译一次成功后再一个个加第三方插件
我就是这样编译成功的，现在和5.10用的插件一致没有报错

jaychin2, 2022-01-02:
我又重拉了一遍，可能是缓存的问题，我还是没成功，重试中

jaychin2, 2022-01-02:
weighted random twos choice least-connection scheduling (IP_VS_TWOS) [N/m/?] (NEW) 
依旧会出现这个选项，到底是哪个位置的这个需要关闭呢？

jaychin2, 2022-01-02:
> 是从头重新配置的吗？拉下来后用默认编译一次5.15的，看行不行，编译一次成功后再一个个加第三方插件 我就是这样编译成功的，现在和5.10用的插件一致没有报错

你有出现选择题吗？
weighted random twos choice least-connection scheduling (IP_VS_TWOS) [N/m/?] (NEW)
我加了三方库之后就出现这个选择题

kleisth, 2022-01-03:
之前编译出错的时候是有这个选择，重新拉源码后就没了。记得要重新make menuconfig，不要把之前的.config复制过来。并且先不加第三方插件，先编译一次，成功后在加。

jaychin2, 2022-01-03:
> 之前编译出错的时候是有这个选择，重新拉源码后就没了。记得要重新make menuconfig，不要把之前的.config复制过来。并且先不加第三方插件，先编译一次，成功后在加。

大佬，我编译成功了，但是还是出现了
weighted random twos choice least-connection scheduling (IP_VS_TWOS) [N/m/?] (NEW)
这个，我选的M，我不知道这个干啥用的，这编译出来的固件会不会有什么问题

WYC-2020, 2022-01-03:
@jaychin2 自己看我提交的issue，我已经解决了并且关闭了

kleisth, 2022-01-03:
> 

> > 之前编译出错的时候是有这个选择，重新拉源码后就没了。记得要重新make menuconfig，不要把之前的.config复制过来。并且先不加第三方插件，先编译一次，成功后在加。
> 
> 大佬，我编译成功了，但是还是出现了 weighted random twos choice least-connection scheduling (IP_VS_TWOS) [N/m/?] (NEW) 这个，我选的M，我不知道这个干啥用的，这编译出来的固件会不会有什么问题
今天我加了docker后也出现这个选择了，一样可以编译成功，使用目前也没发现什么问题

kleisth, 2022-01-03:
> > 之前编译出错的时候是有这个选择，重新拉源码后就没了。记得要重新make menuconfig，不要把之前的.config复制过来。并且先不加第三方插件，先编译一次，成功后在加。
> 
> 大佬，我编译成功了，但是还是出现了 weighted random twos choice least-connection scheduling (IP_VS_TWOS) [N/m/?] (NEW) 这个，我选的M，我不知道这个干啥用的，这编译出来的固件会不会有什么问题

今天我加了docker后也出现这个选择了，一样可以编译成功，使用目前也没发现什么问题

jaychin2, 2022-01-03:
大佬加三方插件已经编译成功了，我想知道哪个是英特尔i210的网卡驱动，以及安装好后怎么开网卡多队列

jaychin2, 2022-01-03:
> > > 之前编译出错的时候是有这个选择，重新拉源码后就没了。记得要重新make menuconfig，不要把之前的.config复制过来。并且先不加第三方插件，先编译一次，成功后在加。
> > 
> > 大佬，我编译成功了，但是还是出现了 weighted random twos choice least-connection scheduling (IP_VS_TWOS) [N/m/?] (NEW) 这个，我选的M，我不知道这个干啥用的，这编译出来的固件会不会有什么问题
> 
> 今天我加了docker后也出现这个选择了，一样可以编译成功，使用目前也没发现什么问题

我去试下是不是docker的问题

jaychin2, 2022-01-03:
> 5.15 不支持 SFE 

大佬加三方插件已经编译成功了，我想知道哪个是英特尔i210的网卡驱动，以及安装好后怎么开网卡多队列

WYC-2020, 2022-01-04:
https://github.com/coolsnowwolf/lede/issues/8567#issue-1091806770
看这个

jaychin2, 2022-01-04:
> https://github.com/coolsnowwolf/lede/issues/8567#issue-1091806770
> 看这个

我试了是docker出那个选择。 大佬你知道哪个是210的网卡驱动吗，我只看到了82576那几个

kleisth, 2022-01-04:
> > [#8567 (comment)](https://github.com/coolsnowwolf/lede/issues/8567#issue-1091806770)
> > 看这个
> 
> 我试了是docker出那个选择。 大佬你知道哪个是210的网卡驱动吗，我只看到了82576那几个

就是igb 82576这个驱动没错啊，默认就包含了的。如果实机安装的话，网卡如果支持多队列的话默认应该也是开启的

jaychin2, 2022-01-04:
> > > [#8567 (comment)](https://github.com/coolsnowwolf/lede/issues/8567#issue-1091806770)
> > > 看这个
> > 
> > 我试了是docker出那个选择。 大佬你知道哪个是210的网卡驱动吗，我只看到了82576那几个
> 
> 就是igb 82576这个驱动没错啊，默认就包含了的。实机安装的话，网卡如果支持多队列的话默认应该也是开启的

好的大佬

jaychin2, 2022-01-04:
> > > [#8567 (comment)](https://github.com/coolsnowwolf/lede/issues/8567#issue-1091806770)
> > > 看这个
> > 
> > 
> > 我试了是docker出那个选择。 大佬你知道哪个是210的网卡驱动吗，我只看到了82576那几个
> 
> 就是igb 82576这个驱动没错啊，默认就包含了的。实机安装的话，网卡如果支持多队列的话默认应该也是开启的

大佬我还有个问题，为什么我编译出来的5.15一直是满频率我cpu是1165G7 2.8GHz，装完固件之后最低就是2800 MHz只高不低

closed 2022-01-09 by jaychin2

wutaijieing, 2023-01-11:
> [#8567 (comment)](https://github.com/coolsnowwolf/lede/issues/8567#issue-1091806770) 看这个

404了。大佬