coolsnowwolf/lede #8582: 5.15选择编译kmod-iwlwifi报错，5.10没问题

lxl6125, 2022-01-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [*] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
/home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c: In function 'iwl_mvm_temp_notif':
/home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:207:3: error: implicit declaration of function 'thermal_notify_framework' [-Werror=implicit-function-declaration]
   thermal_notify_framework(tz_dev->tzone,
   ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[12]: *** [scripts/Makefile.build:277: /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.o] Error 1
make[12]: *** Waiting for unfinished jobs....
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/mesh.o
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/mesh_pathtbl.o
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/mesh_plink.o
make[11]: *** [scripts/Makefile.build:540: /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm] Error 2
make[10]: *** [scripts/Makefile.build:540: /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi] Error 2
make[9]: *** [scripts/Makefile.build:540: /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel] Error 2
make[8]: *** [scripts/Makefile.build:540: /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
make[8]: *** Waiting for unfinished jobs....
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/mesh_hwmp.o
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/mesh_sync.o
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/mesh_ps.o
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/pm.o
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/rc80211_minstrel_ht.o
  CC [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/rc80211_minstrel_ht_debugfs.o
  LD [M]  /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/net/mac80211/mac80211.o
make[7]: *** [Makefile:1868: /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[3]: *** [Makefile:584: /home/lxl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
make[3]: Leaving directory '/home/lxl/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#144.45#15.03#90.08
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/lxl/lede'
make[1]: *** [package/Makefile:110: /home/lxl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lxl/lede'
make: *** [/home/lxl/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-01-08:
https://github.com/WYC-2020/lede/commit/0bdae446b37ca151de2c17902635df59770c5c25 
用这个patch试下

WYC-2020, 2022-01-08:
@lxl6125  用上面我提供的那个patch试下，验证一下功能是否正常

lxl6125, 2022-01-10:
> @lxl6125 用上面我提供的那个patch试下，验证一下功能是否正常

我的是x86_64，用了这个patch正常了

WYC-2020, 2022-01-13:
@lxl6125 ok

coolsnowwolf, 2026-09-06:
旧 backports 5.10 的 `thermal_notify_framework` API 编译错误已随当前 mac80211/backports 更新解决；现行代码不再使用该旧快照，关闭为“已完成”。

closed 2026-09-06 by coolsnowwolf