coolsnowwolf/lede #8583: 5.15 B53错误

vipboddy, 2022-01-02:
make[4]: Entering directory '/home/suling/actions-runner/_work/openwrt-project/openwrt-project/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND objtool
  CHK     include/generated/compile.h
  CC [M]  drivers/net/phy/b53/b53_common.o
drivers/net/phy/b53/b53_common.c: In function 'b53_configure_ports_of':
drivers/net/phy/b53/b53_common.c:528:9: error: this statement may fall through [-Werror=implicit-fallthrough=]
      if (is_imp_port(dev, port_num))
         ^
drivers/net/phy/b53/b53_common.c:533:5: note: here
     case 1000:
     ^~~~
cc1: all warnings being treated as errors
make[8]: *** [scripts/Makefile.build:277: drivers/net/phy/b53/b53_common.o] Error 1
make[7]: *** [scripts/Makefile.build:540: drivers/net/phy/b53] Error 2
make[6]: *** [scripts/Makefile.build:540: drivers/net/phy] Error 2
make[5]: *** [scripts/Makefile.build:540: drivers/net] Error 2
make[4]: *** [Makefile:1868: drivers] Error 2
make[3]: *** [Makefile:28: /home/suling/actions-runner/_work/openwrt-project/openwrt-project/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.modules] Error 2
make[4]: Leaving directory '/home/suling/actions-runner/_work/openwrt-project/openwrt-project/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
make[2]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/suling/actions-runner/_work/openwrt-project/openwrt-project/openwrt/target/linux/x86'
make[2]: Leaving directory '/home/suling/actions-runner/_work/openwrt-project/openwrt-project/openwrt/target/linux'
time: target/linux/compile#5.35#1.26#6.52
    ERROR: target/linux failed to build.
make[1]: *** [target/Makefile:25: target/linux/compile] Error 1
make[1]: Leaving directory '/home/suling/actions-runner/_work/openwrt-project/openwrt-project/openwrt'
make: *** [/home/suling/actions-runner/_work/openwrt-project/openwrt-project/openwrt/include/toplevel.mk:230: target/compile] Error 2
Error: Process completed with exit code 2.

coolsnowwolf, 2022-01-03:
B53 是什么？ 老的 .config ?

xiangfeidexiaohuo, 2022-01-03:
网卡设备选多了，博通的某个

closed 2022-01-04 by AmadeusGhost