coolsnowwolf/lede #8588: 路由器一登录就失联，新版固件。

lonelysoul, 2022-01-03:
路由器一点登录就失联
可以ping通，但ssh无法登陆。
HDMI接显示器查看如图，跑码持续不停，无法结束
[image]
[image]

nb12nb34, 2022-01-03:
同样的问题。。。有谁知道怎么解决吗

zhaoxp520, 2022-01-03:
同样的问题，不用登录，一开机不到两分钟就会这样了，快点登录进去能看到CPU满载。
SSH进去想htop看看~结果SSH进去后基本就卡死了，web后台也进不去了，只能重启。
无奈之下只有搬回房间用PE写了个上个月头编译的固件
放VMware测试也发现一直不停滚动这个速度很快。

pannnnnnn, 2022-01-03:
https://github.com/coolsnowwolf/lede/actions/runs/1638634985 及之前似乎是正常的
https://github.com/coolsnowwolf/lede/actions/runs/1641631847 及之后似乎是有问题的

lxwq, 2022-01-03:
试了下29号编译的,可以正常使用,
31号编译的就一直刷屏了,无奈,用U盘重新刷了系统,

lonelysoul, 2022-01-03:
我用优盘冲刷了好几次了。
无奈，现在去编译官方版本了。
官方版本加helloword，编译成功后点击helloword时候，报错
/usr/lib/lua/luci/dispatcher.lua:1347: module 'luci.cbi' not found:
	no field package.preload['luci.cbi']
	no file './luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi/init.lua'
	no file '/usr/lib/lua/luci/cbi.lua'
	no file '/usr/lib/lua/luci/cbi/init.lua'
	no file './luci/cbi.so'
	no file '/usr/lib/lua/luci/cbi.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:1347: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:1000: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

lxwq, 2022-01-03:
> 我用优盘冲刷了好几次了。 无奈，现在去编译官方版本了。 官方版本加helloword，编译成功后点击helloword时候，报错 /usr/lib/lua/luci/dispatcher.lua:1347: module 'luci.cbi' not found: no field package.preload['luci.cbi'] no file './luci/cbi.lua' no [...]

好像是LUCI版本不兼容,
着急用可以先checkout旧版编译啊,

lonelysoul, 2022-01-03:
一个月前的有一个版本能用，还没启用。
我再编译个带pw的固件，看看pw有没有问题。

lonelysoul, 2022-01-03:
在软件市场装完luci-compat，ssrp正常了。

coolsnowwolf, 2022-01-03:
重新生成 .config 就好了，老配置冲突

zkevinzh, 2022-01-03:
> 重新生成 .config 就好了，老配置冲突

感谢！重新生成后编译就解决了。这个问题折腾我几天了，还以为是我的编译环境问题

bravodilove, 2022-01-03:
> 

我怎么感觉今天更新之后才好的 我昨天 清理了N遍 删除lede文件夹 都删了几回了。。

zkevinzh, 2022-01-03:
> > 
> 
> 我怎么感觉今天更新之后才好的 我昨天 清理了N遍 删除lede文件夹 都删了几回了。。

是的。。我今天上午还把Ubuntu重置过了，编译出来还是不行的。应该是下午l大修复了

bravodilove, 2022-01-03:
> 

我昨天。。编译到凌晨4点多。

WYC-2020, 2022-01-03:
这个应该是陷入无限挂载了

lonelysoul, 2022-01-03:
不知道原因，搞了两天还不行。今天换了原版。

获取 Outlook for iOS<https://aka.ms/o0ukef>

coffee0818, 2022-01-03:
> 不知道原因，搞了两天还不行。今天换了原版。 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

我也是一样的问题，看到L大回复了。引用L大的话：”删除并重新生成 .config 就好了，老配置冲突。“

liubangjian, 2022-01-03:
> > 不知道原因，搞了两天还不行。今天换了原版。 获取 Outlook for iOShttps://aka.ms/o0ukef
> > […](#)
> 
> 我也是一样的问题，看到L大回复了。引用L大的话：”删除并重新生成 .config 就好了，老配置冲突。“

我是用actions来编译的，可以编译成功，但是用在esxi6.7上面就出现上面的情况了。这应该不涉及到旧的config问题吧？每次编译都是新的。

xayane, 2022-01-03:
> > 不知道原因，搞了两天还不行。今天换了原版。 获取 Outlook for iOShttps://aka.ms/o0ukef
> > […](#)
> 
> 我也是一样的问题，看到L大回复了。引用L大的话：”删除并重新生成 .config 就好了，老配置冲突。“

同样的问题，我重新更新了.config文件，还是不行。用action生成的，也不行。

zhaoxp520, 2022-01-03:
> > > 不知道原因，搞了两天还不行。今天换了原版。 获取 Outlook for iOShttps://aka.ms/o0ukef
> > > […](#)
> > 
> > 
> > 我也是一样的问题，看到L大回复了。引用L大的话：”删除并重新生成 .config 就好了，老配置冲突。“
> 
> 同样的问题，我重新更新了.config文件，还是不行。用action生成的，也不行。

更新了今天的源码，make clean 删除.config后重新弄~现在VMware下测试不会了，不过不太敢上机~等没人再更新看看

lxwq, 2022-01-04:
如果要删除.config重建,有点太麻烦了吧,要测试修改很久哦,

lxwq, 2022-01-04:
是不是这个committed引起的?有人测试过没?
https://github.com/coolsnowwolf/lede/commit/01426ea8a080d7786e4f2aaba9a8a69a22fe9388

liubangjian, 2022-01-04:
L大提交新的更新，重新git clone 编译吧！应该是5.15兼容性问题。

lxwq, 2022-01-04:
> L大提交新的更新，重新git clone 编译吧！应该是5.15兼容性问题。

今天编译的还是不行,一直刷屏,
后来把automount取消掉就正常了,

coffee0818, 2022-01-04:
> > > 不知道原因，搞了两天还不行。今天换了原版。 获取 Outlook for iOShttps://aka.ms/o0ukef
> > > […](#)
> > 
> > 
> > 我也是一样的问题，看到L大回复了。引用L大的话：”删除并重新生成 .config 就好了，老配置冲突。“
> 
> 我是用actions来编译的，可以编译成功，但是用在esxi6.7上面就出现上面的情况了。这应该不涉及到旧的config问题吧？每次编译都是新的。

要执行“rm -rf ./tmp && rm -rf .config”，我是一气之下直接rm的lede文件夹，重新clone编译的。目前只有openVPN在Luci界面一直不能启动的问题，用命令行启动就成功了。

coffee0818, 2022-01-04:
> > > 不知道原因，搞了两天还不行。今天换了原版。 获取 Outlook for iOShttps://aka.ms/o0ukef
> > > […](#)
> > 
> > 
> > 我也是一样的问题，看到L大回复了。引用L大的话：”删除并重新生成 .config 就好了，老配置冲突。“
> 
> 同样的问题，我重新更新了.config文件，还是不行。用action生成的，也不行。

不是更新，要删除了rm -rf ./tmp && rm -rf .config，重新用make menuconfig生成，直接保存是有问题的。至少我是重新clone后编译的现在OK了

closed 2022-01-07 by AmadeusGhost