coolsnowwolf/lede #8592: cmGeneratorTarget Error，bootstrapping make错误

chenpengyu119, 2022-01-04:
## 1.关于你要提交的问题

编译报错cmGeneratorTarget 

## 2. 详细叙述

### (1) 具体问题

编译bootstrapping时报错

### (2) 路由器型号和固件版本

orange pi zero plus 
最新固件

### (3) 详细日志

g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://bugzilla.redhat.com/bugzilla> for instructions.
Makefile:182: recipe for target 'cmGeneratorTarget.o' failed
gmake[4]: *** [cmGeneratorTarget.o] Error 4
gmake[4]: *** Waiting for unfinished jobs....
gmake[4]: Leaving directory '/usr/local/tmp/openwrt/lede/build_dir/host/cmake-3.22.1/Bootstrap.cmk'
---------------------------------------------
Error when bootstrapping CMake:
Problem while running gmake
---------------------------------------------
Log of errors: /usr/local/tmp/openwrt/lede/build_dir/host/cmake-3.22.1/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
Makefile:38: recipe for target '/usr/local/tmp/openwrt/lede/build_dir/host/cmake-3.22.1/.configured' failed
make[3]: *** [/usr/local/tmp/openwrt/lede/build_dir/host/cmake-3.22.1/.configured] Error 9
make[3]: Leaving directory '/usr/local/tmp/openwrt/lede/tools/cmake'
time: tools/cmake/compile#347.65#437.58#1318.23
    ERROR: tools/cmake failed to build.
tools/Makefile:158: recipe for target 'tools/cmake/compile' failed
make[2]: *** [tools/cmake/compile] Error 1
make[2]: Leaving directory '/usr/local/tmp/openwrt/lede'
tools/Makefile:156: recipe for target '/usr/local/tmp/openwrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy' failed
make[1]: *** [/usr/local/tmp/openwrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/usr/local/tmp/openwrt/lede'
Build failed - please re-run with -j1 to see the real error message
/usr/local/tmp/openwrt/lede/include/toplevel.mk:228: recipe for target 'world' failed

WYC-2020, 2022-01-04:
万能大法

chenpengyu119, 2022-01-04:
> 

什么万能大法

coolsnowwolf, 2026-09-06:
CMake bootstrap 中 `cc1plus` 被系统 killed，典型是主机内存不足或 OOM，不是源码错误，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf