coolsnowwolf/lede #8596: qbtorrent无法启动。

wkq2004, 2022-01-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
12.31更新源码后编译的固件qbtorrent就无法启动。
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
12.31更新源码后编译的固件qbtorrent就无法启动.
### (1) 具体问题

A：

### (2) 路由器型号和固件版X86
A：X86X64 5.10.89

### (3) 详细日志

A：

lxl6125, 2022-01-04:
用 Static Build 试试，我用 Dynamic Build 也是无法启动，换成 Static Build 就没事了

ShinyAsuna, 2022-01-08:
arm 软路由同样报错错误日志同lz

musclemm, 2022-01-10:
我编译的时候选择 Static Build ，但是会报错，选择 Dynamic Build的时候可以编译成功。

check_data_file_clashes: Package qBittorrent-static wants to install file /home/ubuntu/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/qbittorrent-nox
        But that file is already provided by package  * qbittorrent
 * opkg_install_cmd: Cannot install package qBittorrent-static.

> Build Version Selection 用 Static Build 试试，我用 Dynamic Build 也是无法启动，换成 Static Build 就没事了

coolsnowwolf, 2026-09-06:
qBittorrent crash loop 没有程序版本、依赖库或崩溃日志，无法定位，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf