coolsnowwolf/lede #8598: 5.15内核的ntfs3驱动问题。

flyhong666, 2022-01-04:
[image]
5.15内核下的ntfs3驱动不会自动加载ntfs格式分区到已挂载点。启用下面的那个挂载点后，共享出来。其是采用rom空余空间大小来拷贝文件的，随便拷贝一个1G以上文件就会提示空间不足，请问这个该如何处理？5.4以及5.10内核采用的antfs却一切正常没有这种情况。

flyhong666, 2022-01-04:
[image]
空间不够。。。。

WYC-2020, 2022-01-04:
不会吧？明天我编译一下测试看看

flyhong666, 2022-01-04:
> 不会吧？明天我编译一下测试看看

是的。ksmbd以及samba 我都试过了。都是这样的毛病。

flyhong666, 2022-01-04:
5.10 会自动挂载上去。。。
[image]
[image]

flyhong666, 2022-01-05:
> 不会吧？明天我编译一下测试看看

哥，编译试过没？

WYC-2020, 2022-01-05:
@flyhong666 我这边测试了，确实不会自动挂载，只是我用的ext4 这个格式测试的，哪个smab共享直接不行，ksmb这个可以，并且没出现你说的哪个使用rom空间的问题，我8G我考了一个7G文件都没问题

flyhong666, 2022-01-05:
> @flyhong666 我这边测试了，确实不会自动挂载，只是我用的ext4 这个格式测试的，哪个smab共享直接不行，ksmb这个可以，并且没出现你说的哪个使用rom空间的问题，我8G我考了一个7G文件都没问题

ntfs 格式有问题，ext4的没事。我在虚拟机上测试过。。。你要不整个ntfs格式的测试下。

WYC-2020, 2022-01-05:
@flyhong666 我都直接挂载不上ntfs未知文件系统哈哈，不折腾了，懒得搞5.15，问题贼多，等我要用到了再折腾

flyhong666, 2022-01-05:
> @flyhong666 我都直接挂载不上ntfs未知文件系统哈哈，不折腾了，懒得搞5.15，问题贼多，等我要用到了再折腾

你看我第一张图的挂载点那里，可以识别，估计是挂载不上去导致的。

flyhong666, 2022-01-05:
看了下启动日志，还真的是挂载失败导致。。。。

Wed Jan  5 17:03:35 2022 daemon.err block: No "mount.ntfs" utility available
Wed Jan  5 17:03:35 2022 daemon.err block: mounting /dev/sdc1 (ntfs) as /mnt/sdc1 failed (25) - Not a tty

WYC-2020, 2022-01-05:
经过我的再一次验证，和修改 现在可以挂载了但是也只能手动挂载 并且共享访问 没问题 用的ksmbd ，samba我这边始终没法共享，再看下再说

flyhong666, 2022-01-05:
> 经过我的再一次验证，和修改 现在可以挂载了但是也只能手动挂载 并且共享访问 没问题 用的ksmbd ，samba我这边始终没法共享，再看下再说

哥，怎么处理的，指点下。。。

WYC-2020, 2022-01-05:
https://github.com/WYC-2020/lede/commit/c2690c17b1e16514ae0148d5b904cf70f843c6f0

flyhong666, 2022-01-05:
> https://github.com/WYC-2020/lede/commit/c2690c17b1e16514ae0148d5b904cf70f843c6f0

提交更新到大雕代码那去。

flyhong666, 2022-01-06:
5.15内核的ntfs3驱动，还真的是有质的飞跃。。。挂了一个ntfs格式的硬盘在路由器，以前写能达到100M，读只能30-40M。但现在读可以达到90M 了。。。猛啊。。。
[image]
[image]

closed 2022-01-07 by AmadeusGhost