coolsnowwolf/lede #8599: 5.15版本NFS还是不能编译

zd911222, 2022-01-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译时在nfs处出错，提示缺少nfs_ssc.ko

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

A：ERROR: module '/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/fs/nfs_common/nfs_ssc.ko' is missing.
make[3]: *** [modules/fs.mk:424: /home/zd/lede/bin/targets/x86/64/packages/kmod-fs-nfs-common_5.15.12-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zd/lede/package/kernel/linux'
time: package/kernel/linux/compile#4.48#0.68#4.41
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zd/lede'
make[1]: *** [package/Makefile:110: /home/zd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zd/lede'
make: *** [/home/zd/lede/include/toplevel.mk:230：world] 错误 2

flyhong666, 2022-01-04:
你这是启用了啥组件？取消那个呗。

WYC-2020, 2022-01-04:
怕又是在5.15以前的config 上配置的吧

flyhong666, 2022-01-04:
> 怕又是在5.15以前的config 上配置的吧

哥，修一下你的那个ksmbd 呗。

WYC-2020, 2022-01-04:
修什么？没问题啊我用起 @flyhong666

flyhong666, 2022-01-04:
> 修什么？没问题啊我用起 @flyhong666

用你的文件覆盖大雕的，真是楼主出现的错误信息。。

Gzxhwq, 2022-01-04:
重新配置的.config，丢action上关cache编译也是一样的报错

WYC-2020, 2022-01-04:
@flyhong666 你用什么覆盖什么哦，都不懂你说的啥，明天我编译看下吧

GreatMichaelLee, 2022-01-04:
> 重新配置的.config，丢action上关cache编译也是一样的报错

我都不敢提这个问题了，上次提了，而且我也说我重新make menuconfig了还是有，结果大雕一上来就直接给我close了。到现在我也是报这个错，没法编译5.15，只能5.10.89

GreatMichaelLee, 2022-01-04:
bnx2-firmware/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.bnx2-firmware_installed
echo "bnx2-firmware" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/builder/lede_x86/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#0.39#2.99#2.24
make[3]: Entering directory '/home/builder/lede_x86/package/firmware/prism54-firmware'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/builder/lede_x86/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.21#0.05#0.24
make[3]: Entering directory '/home/builder/lede_x86/package/kernel/linux'
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
ERROR: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/fs/nfs_common/nfs_ssc.ko' is missing.
make[3]: *** [modules/fs.mk:424: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fs-nfs-common_5.15.12-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#2.49#0.29#2.77
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

WYC-2020, 2022-01-04:
明天给我一份你们编译选了那些插件，我编译看下

flyhong666, 2022-01-04:
> bnx2-firmware/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/' touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.bnx2-firmware_installed echo [...]

可能你用的组件调用到那个kmod-fs-nfs-common 了。。。。

GreatMichaelLee, 2022-01-04:
> 可能你用的组件调用到那个kmod-fs-nfs-common 了。。。。

那要怎么搞？

GreatMichaelLee, 2022-01-04:
> 明天给我一份你们编译选了那些插件

> 明天给我一份你们编译选了那些插件，我编译看下

麻烦大佬看看
[config.zip](https://github.com/coolsnowwolf/lede/files/7808424/config.zip)

flyhong666, 2022-01-04:
> > 明天给我一份你们编译选了那些插件
> 
> > 明天给我一份你们编译选了那些插件，我编译看下
> 
> 麻烦大佬看看 [config.zip](https://github.com/coolsnowwolf/lede/files/7808424/config.zip)

我不懂怎么看这配置里面那个东西调用到了这个依赖。

Gzxhwq, 2022-01-04:
@WYC-2020 
https://github.com/Gzxhwq/OpenWrt-Actions/blob/18ad6a7064b0e7da5389aa7cf9ebe20e48de9f04/x86_64.config

xiangfeidexiaohuo, 2022-01-04:
不敢提，nfs_ssc也启用了，反正编译这个报错一直

GreatMichaelLee, 2022-01-05:
> 不敢提，nfs_ssc也启用了，反正编译这个报错一直

是的，不敢提，提了就是被关。。。例如#8572, 只好混在回帖里，抖抖索索的说几句-_-!

closed 2022-01-05 by AmadeusGhost

WYC-2020, 2022-01-05:
@GreatMichaelLee  你这个真的是，恨不得把所有插件都选了啊，有的插件会冲突的，我先编译看下吧