coolsnowwolf/lede #8601: 编译出错

gswangmd-gssi, 2022-01-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-mmc into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-mmc_5.10.89-1_x86_64.ipk
NOTICE: module '/home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/drivers/mmc/host/sdhci.ko' is built-in.
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sdhci/lib/modules/5.10.89/sdhci-pltfm.ko: relocatable
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sdhci into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-sdhci_5.10.89-1_x86_64.ipk
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-thermal into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-thermal_5.10.89-1_x86_64.ipk
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-mmc-spi/lib/modules/5.10.89/mmc_spi.ko: relocatable
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-mmc-spi into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-mmc-spi_5.10.89-1_x86_64.ipk
NOTICE: module '/home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/drivers/usb/core/usbcore.ko' is built-in.
NOTICE: module '/home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/drivers/usb/common/usb-common.ko' is built-in.
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-core into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-usb-core_5.10.89-1_x86_64.ipk
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-printer/lib/modules/5.10.89/usblp.ko: relocatable
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-printer into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-usb-printer_5.10.89-1_x86_64.ipk
NOTICE: module '/home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/drivers/usb/storage/usb-storage.ko' is built-in.
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-usb-storage_5.10.89-1_x86_64.ipk
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-jumpshot.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-freecom.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-usbat.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-karma.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-alauda.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-cypress.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-sddr09.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-sddr55.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-datafab.ko: relocatable
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras/lib/modules/5.10.89/ums-isd200.ko: relocatable
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-extras into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-usb-storage-extras_5.10.89-1_x86_64.ipk
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-uas/lib/modules/5.10.89/uas.ko: relocatable
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-storage-uas into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-usb-storage-uas_5.10.89-1_x86_64.ipk
rstrip.sh: /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-net/lib/modules/5.10.89/usbnet.ko: relocatable
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-net into /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-usb-net_5.10.89-1_x86_64.ipk
ERROR: module '/home/gswangmd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/net/core/selftests.ko' is missing.
make[3]: *** [modules/usb.mk:1154: /home/gswangmd/lede/bin/targets/x86/64/packages/kmod-usb-net-asix_5.10.89-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/gswangmd/lede/package/kernel/linux'
time: package/kernel/linux/compile#10.95#0.78#9.59
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [package/Makefile:110: /home/gswangmd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:230: world] Error 2

zhangguanzhang, 2022-01-05:
看样子是 `kmod-usb-net-asix` 被改出问题了

dengchan0416, 2022-01-05:
我也是遇到这个问题

closed 2022-01-05 by AmadeusGhost