coolsnowwolf/lede #8604: 请问云编译,怎么修改选择内核版本编译

ayancn, 2022-01-05:
是修改.config文件还是openwrt.yml
修改.config文件增加内核代码能否实现
想要编译5.15版内核,想要修改什么文件,增加什么代码,希望修改.config文件就能实现

WYC-2020, 2022-01-05:
纯小白玩编译哈4

zhangguanzhang, 2022-01-05:
```
CONFIG_HAS_TESTING_KERNEL=y
```

Michaol, 2022-01-05:
> ```
> CONFIG_HAS_TESTING_KERNEL=y
> ```

CONFIG_LINUX_5_15=y
别漏了这个

Michaol, 2022-01-05:
要么云编译开SSH，make menuconfig选择testing kernel。要么手搓config。

ayancn, 2022-01-05:
> 要么云编译开SSH，make menuconfig选择testing kernel。要么本地make menuconfig后扔上去。要么手搓config。

.config配置文件增加这3行吗
CONFIG_HAS_TESTING_KERNEL=y
CONFIG_TESTING_KERNEL=y
CONFIG_LINUX_5_15=y

还是只要增加下面两行

CONFIG_TESTING_KERNEL=y
CONFIG_LINUX_5_15=y

Michaol, 2022-01-05:
> 

建议你本地重新建config，否则大几率失败。我就是懒，失败了一天，最后乖乖重建config就好了。

CONFIG_HAS_TESTING_KERNEL=y默认config本身有，不需要改，除非你改过。其他两个config里面都有，不用加。CONFIG_TESTING_KERNEL默认not set。CONFIG_LINUX_5_10=y默认，改为CONFIG_LINUX_5_15=y。

ayancn, 2022-01-05:
> > 
> 
> 建议你本地重新建config，否则大几率失败。我就是懒，失败了一天，最后乖乖重建config就好了。
> 
> CONFIG_HAS_TESTING_KERNEL=y默认config本身有，不需要改，除非你改过。其他两个config里面都有，不用加。CONFIG_TESTING_KERNEL默认not set。CONFIG_LINUX_5_10=y默认，改为CONFIG_LINUX_5_15=y。

> CONFIG_HAS_TESTING_KERNEL=y
> CONFIG_TESTING_KERNEL=y
> CONFIG_LINUX_5_15=y

没有安装linux 只使用云编译, 业余编译方便自用 所以只使用云编译

在目前正常使用成功编译的.config文件中加入什么代码可以编译5.15.12版本的?

pcvc, 2022-01-05:
听说 5.15 解决了 i225 网卡驱动的问题？ @ayancn

ayancn, 2022-01-05:
试了,目前 ramips/ MediaTek MT7621  / Xiaomi Mi Router CR660x 只能编译最新的是5.10.89 设置5.15也是编译5.10,可能库中还没添加5.15内核支持?

pcvc, 2022-01-05:
我刚刚下单买了J4125 i225-V网卡的，就怕到时固件不兼容，先了解一下 @ayancn

xyskwjy, 2022-01-06:
sh1加个命令。然后，5.10其实也支持225网卡
sed -i 's/KERNEL_PATCHVER:=5.10/KERNEL_PATCHVER:=5.15/g' ./target/linux/x86/Makefile

closed 2022-01-06 by AmadeusGhost

osmboy, 2023-01-18:
小版本如何选择？

choto-silagen, 2023-05-12:
在/lede/target/linux目录下找到你需要编译的系统文件夹，例如x86,点击文件夹后，找到Makefile双击文本修改这一行：
KERNEL_TESTING_PATCHVER:5.15
改成你需要的内核，例如：KERNEL_TESTING_PATCHVER:6.1
然后保存并关闭文档

LIUBANGJIAN, 2024-01-24:
> > 
> 
> 建议你本地重新建config，否则大几率失败。我就是懒，失败了一天，最后乖乖重建config就好了。
> 
> CONFIG_HAS_TESTING_KERNEL=y默认config本身有，不需要改，除非你改过。其他两个config里面都有，不用加。CONFIG_TESTING_KERNEL默认not set。CONFIG_LINUX_5_10=y默认，改为CONFIG_LINUX_5_15=y。

我现在想编译6.6的内核，添加上述设置，并把CONFIG_LINUX_6_6=y，并没有效果，请问如何解决？我本地按照L大的命令编译也没找到testing kernel的选择，是我没有找到吗？请大佬们指导一下。