coolsnowwolf/lede #8608: 升级后DNS服务出错

wang1zhen, 2022-01-05:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：重新编译新版本固件后，DNS服务出错，无法正确解析

### (2) 路由器型号和固件版本

A：X86，使用最新源码编译，加入了TurboAcc的DNS缓存功能

WYC-2020, 2022-01-05:
turbo的缓存功能基本废了，都已经被我精简掉了 你居然还在用那玩意的dns缓存

wang1zhen, 2022-01-05:
> turbo的缓存功能基本废了，都已经被我精简掉了 你居然还在用那玩意的dns缓存

删除turboacc之后仍然保持了相同的问题，应该不是turboacc导致的

wang1zhen, 2022-01-05:
补充系统日志：
```
Wed Jan  5 14:31:42 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:31:42 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:31:47 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:31:47 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:31:51 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:31:51 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:32:56 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:32:56 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:33:00 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:33:00 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:34:05 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:34:05 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:34:09 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:34:09 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:35:14 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:35:14 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:35:18 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:35:18 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:36:24 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:36:24 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:36:27 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:36:27 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:37:32 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:37:32 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:37:36 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:37:36 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:38:41 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:38:41 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:38:45 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:38:45 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:39:50 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:39:50 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:39:54 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:39:54 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jan  5 14:40:12 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jan  5 14:40:12 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
```

`nslookup`输出：
```
PS C:\Users\wang1> nslookup.exe baidu.com
服务器:  UnKnown
Address:  240e:389:8800:a700::1

*** UnKnown 找不到 baidu.com: Query refused
```

然而使用openwrt网页端的网络诊断，发现nslookup是成功的

weilai-all, 2022-01-05:
碰到过类似情况，x86 5.15.12， https://github.com/coolsnowwolf/lede/commit/7b75725731c013f871b6bb718e03a0e580483328#comments 之后就有这个问题了。

minlang112, 2022-01-07:
> 补充系统日志：
>
> ```
> Wed Jan  5 14:31:42 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
> Wed Jan  5 14:31:42 2022 user.notice dnsmasq: Allowing [...]

并且我发现一个奇异的现象，这个版本+ssp的GFW模式似乎失效了，用绕过中国大陆模式一切正常，切到GFW模式就完犊子。

firkerword, 2022-01-09:
[image]
我也遇到这样的问题   dns跟之前设置一样 最近才出现这种情况的  切换了很多种dns设置都是这样

closed 2022-06-10 by wang1zhen