coolsnowwolf/lede #8610: 1.5 commit 5.15报错 antfs

GreatMichaelLee, 2022-01-05:
单线程build  x86 kernel 5.15 build

/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1425:13: note: (near initialization for 'antfs_common_inode_operations.getattr')
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1441:13: error: initialization of 'int (*)(struct user_namespace *, struct dentry *, struct iattr *)' from incompatible pointer type 'int (*)(struct dentry *, struct iattr *)' [-Werror=incompatible-pointer-types]
  .setattr = antfs_setattr,
             ^~~~~~~~~~~~~
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1441:13: note: (near initialization for 'antfs_symlink_inode_operations.setattr')
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1456:13: error: initialization of 'int (*)(struct user_namespace *, const struct path *, struct kstat *, u32,  unsigned int)' {aka 'int (*)(struct user_namespace *, const struct path *, struct kstat *, unsigned int,  unsigned int)'} from incompatible pointer type 'int (*)(const struct path *, struct kstat *, u32,  unsigned int)' {aka 'int (*)(const struct path *, struct kstat *, unsigned int,  unsigned int)'} [-Werror=incompatible-pointer-types]
  .getattr = antfs_getattr,
             ^~~~~~~~~~~~~
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1456:13: note: (near initialization for 'antfs_symlink_inode_operations.getattr')
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:277: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.o] Error 1
make[4]: *** [Makefile:1868: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
make[3]: *** [Makefile:47: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/kernel/antfs'
time: package/feeds/packages/antfs/compile#0.77#0.30#1.08
    ERROR: package/feeds/packages/antfs failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/antfs/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$
builder@Build-Server:~/lede_x86$

WYC-2020, 2022-01-05:
函数原型不匹配，你怕是自己修改过吧，我记得5.15是不会使用antfs

GreatMichaelLee, 2022-01-05:
> 函数原型不匹配，你怕是自己修改过吧，我记得5.15是不会使用antfs

没有啊。。。

WYC-2020, 2022-01-05:
@GreatMichaelLee 那看来又是你选了什么包依赖这个了

GreatMichaelLee, 2022-01-05:
> @GreatMichaelLee 那看来又是你选了什么包依赖这个了

同样的配置，不勾选testing kernel, 用5.10.89 一点问题都没有，这个build server我就基本没动过config

WYC-2020, 2022-01-05:
5.10 和5.15 不能再原来的config上面修改 ，比如吧原来的config删除重新配置 5.15 不能兼容5.10 5.4的config 你试下看 @GreatMichaelLee

xiangfeidexiaohuo, 2022-01-05:
5.15不使用antfs，你去掉antfs吧，用的ntfs3啊

GreatMichaelLee, 2022-01-05:
> 5.15不使用antfs，你去掉antfs吧，用的ntfs3啊

我试试，一会反馈

GreatMichaelLee, 2022-01-05:
> 5.15不使用antfs，你去掉antfs吧，用的ntfs3啊

根本没法去掉antfs啊，不让选@xiangfeidexiaohuo

[image]

flyhong666, 2022-01-05:
> > 5.15不使用antfs，你去掉antfs吧，用的ntfs3啊
> 
> 根本没法去掉antfs啊，不让选@xiangfeidexiaohuo
> 
> [image]

先将utilies-filesystem 里面的那个antfs-mount 取消然后就可以取消你这个了。别折腾5.15 了。还不成熟。

GreatMichaelLee, 2022-01-05:
> > > 5.15不使用antfs，你去掉antfs吧，用的ntfs3啊
> > 
> > 
> > 根本没法去掉antfs啊，不让选@xiangfeidexiaohuo
> > [image]
> 
> 先将utilies-filesystem 里面的那个antfs-mount 取消然后就可以取消你这个了。别折腾5.15 了。还不成熟。

有道理，我决定先回5.10.x

WYC-2020, 2022-01-05:
修复完毕，等明天验证了5.4给你们patch

WYC-2020, 2022-01-05:
你这个我遇到了 是你第一次选了不是5.15的内核 然后保存了然后你有重新选择了5.15，这时config里面会有非5.15的模块，简单方法是删除.config ,然后重新配置一下
这个是修复不能自动挂载ntfs的https://github.com/WYC-2020/lede/commit/c2690c17b1e16514ae0148d5b904cf70f843c6f0

GreatMichaelLee, 2022-01-06:
> 你这个我遇到了 是你第一次选了不是5.15的内核 然后保存了然后你有重新选择了5.15，这时config里面会有非5.15的模块，简单方法是删除.config ,然后重新配置一下 这个是修复不能自动挂载ntfs的[WYC-2020/lede@c2690c1](https://github.com/WYC-2020/lede/commit/c2690c17b1e16514ae0148d5b904cf70f843c6f0)

5.10.x最新的code已经可以了，但5.15还是不行，还是antfs报错，前面有兄弟说5.15不用antfs

GreatMichaelLee, 2022-01-06:
> > 你这个我遇到了 是你第一次选了不是5.15的内核 然后保存了然后你有重新选择了5.15，这时config里面会有非5.15的模块，简单方法是删除.config ,然后重新配置一下 这个是修复不能自动挂载ntfs的[WYC-2020/lede@c2690c1](https://github.com/WYC-2020/lede/commit/c2690c17b1e16514ae0148d5b904cf70f843c6f0)
> 
> 5.10.x最新的code已经可以了，但5.15还是不行，还是antfs报错，前面有兄弟说5.15不用antfs

晕死了，按照上面兄弟说的把antfs关了，结果5.15下面又报nfs_ssc.ko丢失，不知道那个fix为啥要加个@5.10.ge?

xiangfeidexiaohuo, 2022-01-06:
> 晕死了，按照上面兄弟说的把antfs关了，结果5.15下面又报nfs_ssc.ko丢失，不知道那个fix为啥要加个@5.10.ge?

你要用5.15 把这个d9316c5814b1f487ebd9ed4df1d2b4263b79f720 去掉呗

closed 2022-01-07 by AmadeusGhost