coolsnowwolf/lede #8611: 最新代理编译一直失败，make distclean后重新编译还是不行

Yofo-me, 2022-01-05:
### log
```shell
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-bcma into /home/joky/lede/bin/targets/x86/64/packages/kmod-bcma_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ikconfig/lib/modules/5.10.89/configs.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ikconfig into /home/joky/lede/bin/targets/x86/64/packages/kmod-ikconfig_5.10.89-1_x86_64.ipk
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-thermal into /home/joky/lede/bin/targets/x86/64/packages/kmod-thermal_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd-rawmidi.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd-pcm.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd-mixer-oss.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd-compress.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd-pcm-oss.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd-seq-device.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd-timer.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/soundcore.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core/lib/modules/5.10.89/snd-hwdep.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-core into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-core_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ac97/lib/modules/5.10.89/snd-ac97-codec.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ac97/lib/modules/5.10.89/ac97_bus.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ac97 into /home/joky/lede/bin/targets/x86/64/packages/kmod-ac97_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-mpu401/lib/modules/5.10.89/snd-mpu401-uart.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-mpu401 into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-mpu401_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-i8x0/lib/modules/5.10.89/snd-intel8x0.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-i8x0 into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-i8x0_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-via82xx/lib/modules/5.10.89/snd-via82xx.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-via82xx into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-via82xx_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-core/lib/modules/5.10.89/snd-hda-core.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-core/lib/modules/5.10.89/snd-hda-codec-generic.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-core/lib/modules/5.10.89/snd-hda-codec.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-core into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-hda-core_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-codec-realtek/lib/modules/5.10.89/snd-hda-codec-realtek.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-codec-realtek into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-hda-codec-realtek_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-codec-via/lib/modules/5.10.89/snd-hda-codec-via.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-codec-via into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-hda-codec-via_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-codec-hdmi/lib/modules/5.10.89/snd-hda-codec-hdmi.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-codec-hdmi into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-hda-codec-hdmi_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-intel/lib/modules/5.10.89/snd-intel-dspcfg.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-intel/lib/modules/5.10.89/snd-hda-intel.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-sound-hda-intel into /home/joky/lede/bin/targets/x86/64/packages/kmod-sound-hda-intel_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-mmc-spi/lib/modules/5.10.89/mmc_spi.ko: relocatable
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-mmc-spi into /home/joky/lede/bin/targets/x86/64/packages/kmod-mmc-spi_5.10.89-1_x86_64.ipk
NOTICE: module '/home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/drivers/usb/host/ohci-hcd.ko' is built-in.
NOTICE: module '/home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/drivers/usb/host/ohci-platform.ko' is built-in.
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-ohci into /home/joky/lede/bin/targets/x86/64/packages/kmod-usb-ohci_5.10.89-1_x86_64.ipk
NOTICE: module '/home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/drivers/usb/host/ohci-pci.ko' is built-in.
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-ohci-pci into /home/joky/lede/bin/targets/x86/64/packages/kmod-usb-ohci-pci_5.10.89-1_x86_64.ipk
NOTICE: module '/home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.89/drivers/usb/host/ehci-hcd.ko' is built-in.
Packaged contents of /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-ehci into /home/joky/lede/bin/targets/x86/64/packages/kmod-usb-ehci_5.10.89-1_x86_64.ipk
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/5.10.89/ehci-platform.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/5.10.89/fsl-mph-dr-of.ko: relocatable
rstrip.sh: /home/joky/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/5.10.89/ehci-fsl.ko: relocatable
[... cut ...]
make[3]: Leaving directory '/home/joky/lede/package/kernel/linux'
time: package/kernel/linux/compile#28.22#5.54#29.04
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/joky/lede'
make[1]: *** [package/Makefile:110: /home/joky/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/joky/lede'
make: *** [/home/joky/lede/include/toplevel.mk:230: world] Error 2
```

sbwcwfhv001, 2022-01-05:
我昨天都成功了，你可能是有插件出错了。

OdysseyLinux, 2022-01-05:
我也是这样，插件没动，默认的几个插件

Django9264, 2022-01-05:
> 我昨天都成功了，你可能是有插件出错了。

我昨天成功了，今天也是失败

WYC-2020, 2022-01-05:
这个已经修复了啊

Yofo-me, 2022-01-06:
> 我昨天都成功了，你可能是有插件出错了。

没有啊，同一个配置文件，2022-1-4号还能编译成功，1月5号我git pull后编译就失败，如log所示；`make distclean`后重新编译之前的配置文件也是这样的错误

Yofo-me, 2022-01-06:
> 这个已经修复了啊

我重新试试

closed 2022-01-07 by AmadeusGhost