coolsnowwolf/lede #8613: 5.10.89的SSP是不是有问题啊

jt8823, 2022-01-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我有两个VPS 相同方式搭建的V2RAY
今天重新编译了一个固件  其中一个V2RAY在openwrt里用不了了
在小飞机里两个都是正常的

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

mitsukileung, 2022-01-05:
[luci-app-ssr-plus: remove alterId config ](https://github.com/fw876/helloworld/commit/cb41416363363938b78a3bc89410fdc0c8164860)

jt8823, 2022-01-05:
> 应该是和这个有关吧 [luci-app-ssr-plus： remove alterId config](https://github.com/fw876/helloworld/commit/cb41416363363938b78a3bc89410fdc0c8164860)

看不懂这个啊   有办法解决么  大佬

EnnawYang, 2022-01-05:
> > 应该是和这个有关吧 [luci-app-ssr-plus： remove alterId config](https://github.com/fw876/helloworld/commit/cb41416363363938b78a3bc89410fdc0c8164860)
> 
> 看不懂这个啊 有办法解决么 大佬

https://github.com/fw876/helloworld/issues/758#issuecomment-1003808048

flyhong666, 2022-01-05:
将你后台的alterid 改成 0 .。。。

jt8823, 2022-01-05:
> 将你后台的alterid 改成 0 .。。。

有办法回退v2ray 老版本吗

Dimitri2020007, 2022-01-06:
xray从2022.1.1取消了vmess的alterid

服务端也要去掉alterid

使用v2ray把alterid改成0就没事了

mitsukileung, 2022-01-06:
> > 将你后台的alterid 改成 0 .。。。
> 
> 有办法回退v2ray 老版本吗

自己手动修改吧
按照[ 这里 ](https://github.com/fw876/helloworld/commit/cb41416363363938b78a3bc89410fdc0c8164860)手动把红色的部份添加到对应的文件里，要修改4个文件

jt8823, 2022-01-06:
> > > 将你后台的alterid 改成 0 .。。。
> > 
> > 
> > 有办法回退v2ray 老版本吗
> 
> 自己手动修改吧 按照[ 这里 ](https://github.com/fw876/helloworld/commit/cb41416363363938b78a3bc89410fdc0c8164860)手动把红色的部份添加到对应的文件里，要修改4个文件

噢  谢了  这样就明白了
 luci-app-ssr-plus 这个的路径是啥啊!

mitsukileung, 2022-01-06:
> 
lede/feeds/helloworld

closed 2022-01-07 by AmadeusGhost