coolsnowwolf/lede #8614: goproxy.baidu证书过期

qd201211, 2022-01-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译时会提示TLS问题，应该是https://goproxy.baidu.com证书过期了，我现在换成https://goproxy.cn可以正常编译通过

### (2) 路由器型号和固件版本

A：自用X86，最新源码

### (3) 详细日志

A：certificate has expired or is not yet valid: current time xxxxxx

CuteTony, 2022-01-05:
过期很久了，换一个就完事了。https://goproxy.io 也可以正常使用。

qd201211, 2022-01-06:
> 过期很久了，换一个就完事了。https://goproxy.io 也可以正常使用。

太闹了，睡了一觉全是这报错。哎，看也没同类型issues我还以为只有我这样呢

CuteTony, 2022-02-12:
To Whom It May Concern:This is an automatic reply, confirming that your E-mail was received.This E-mail (and any attachments or hyperlinks within it) will not be delivered until it passes the anti-virus check by TLAN Project.If you want to send files to me directly, simply click File Collect and follow the instructions.I offer my apology to you for the inconvenience has caused you.Best Regards.Tony  Guo